commaai/openpilot #31703: Update 0.9.6

amd006, 2024-03-05:
### Describe the bug

My Comma 3x is no longer working. AND I keep getting an engine check light on my 2019 Subaru Crosstrek display since February 28th.

### Provide a route where the issue occurs

What

### openpilot version

0.9.6

### Additional info

No

sshane, 2024-03-05:
What is the dongle id your device shows in the settings?

amd006, 2024-03-05:
f45981c50f933ecd

amd006, 2024-03-05:
Has this happened to anyone else?

sshane, 2024-03-05:
This started happening to you before you updated to 0.9.6, I don't see any routes not from 0.9.5 with cruise faulted.

Since nothing software-wise has changed (git commit, version, diff), I don't see anything wrong different about the ECU FW, no messages are getting blocked, and panda is reporting no errors, can you get a route with **openpilot toggled off**? There is a chance that the camera is faulting on its own.

sshane, 2024-03-05:
@jnewb1 can we log ACC faults? It wasn't clear which message to check since there's a few `Cruise_Fault` signals, but they were 1 on this user's drives so it appears they should be checked.

jnewb1, 2024-03-05:
tried here: https://github.com/commaai/openpilot/pull/30546, but `Cruise_Fault` goes high while the camera is initializing so we'd have to have some kind of logic to only show the message after some period of time. I can do that

sshane, 2024-03-05:
There's a few, did you check them all?

jnewb1, 2024-03-05:
yea I checked a few IIRC, `ES_DashStatus` is the one for the dashlights, and then most of the other ES_* signals have a cruise_fault signal as well. I'll look some more

amd006, 2024-03-06:
It works now! Thanks!

closed 2024-03-06 by sshane