commaai/openpilot #31705: Toyota：2024 RAV4 show "steering temporarily unavailable" at low speed

Saber422, 2024-03-05:
### Describe the bug

Test 2024 RAV4 disable radar function and longdutinal control.
OP show "steering temporarily unavailable" for a while and gone below 30kmh occasionally.

### Which car does this affect?

Toyota 2024 RAV4 Taiwan Version

### Provide a route where the issue occurs

0bbe367c98fa1538/2024-03-05--10-51-30/0

### openpilot version

0.9.7 master

Saber422, 2024-03-05:
There is a problem, I am not in standard openpilot, I modify MIN_LATERAL_CONTROL_SPEED = 2.0  # m/s because of my car.
I will ask owner give me more clean data to you.

sshane, 2024-03-05:
Can you send a connect route marked exactly where the fault occurs? Is it only after overriding? The Toyota high angle rate fault does not currently work since the message rate is different.

dennisshen7, 2024-04-03:
I think I have the logs here
https://connect.comma.ai/ff98279bde75c398/1710500382318/1710501265850

Please feel free to let me know for any further help.

Thanks

sshane, 2024-04-17:
Do you have any cases with full rlogs uploaded?

sshane, 2024-04-17:
We don't generally look into issues on forks (this one has MADS it looks like), but this temporary fault is expected at this time. The fault is caused by the prolonged high angle rate:

[image]

We've worked around this with torque control cars, but because the fix isn't directly applicable (different steering msg rates and the fix doesn't improve much due to the low angle rate limits), we aren't prioritizing it at this time.

closed 2024-04-17 by sshane