commaai/openpilot #31708: ONNX reports CUDAExecutionProvider as available even when it is not

mimi89999, 2024-03-05:
### Describe the bug

Hello,
I noticed that ONNX reports `CUDAExecutionProvider` as available even when it is not (like on my Intel GPU).
```
>>> import onnxruntime as ort
>>> print(ort.get_available_providers())
['TensorrtExecutionProvider', 'CUDAExecutionProvider', 'AzureExecutionProvider', 'CPUExecutionProvider']
```

This is not a problem because ONNX automatically falls back to `CPUExecutionProvider`
```
Onnx selected provider:  [('CUDAExecutionProvider', {'cudnn_conv_algo_search': 'DEFAULT'})]
2024-03-05 11:40:14.847702004 [E:onnxruntime:Default, provider_bridge_ort.cc:1548 TryGetProviderInfo_CUDA] /onnxruntime_src/onnxruntime/core/session/provider_bridge_ort.cc:1209 onnxruntime::Provider& onnxruntime::ProviderLibrary::Get() [ONNXRuntimeError] : 1 : FAIL : Failed to load library libonnxruntime_providers_cuda.so with error: libcublasLt.so.11: cannot open shared object file: No such file or directory

2024-03-05 11:40:14.847815005 [W:onnxruntime:Default, onnxruntime_pybind_state.cc:861 CreateExecutionProviderInstance] Failed to create CUDAExecutionProvider. Please reference https://onnxruntime.ai/docs/execution-providers/CUDA-ExecutionProvider.html#requirementsto ensure all dependencies are met.
Onnx using  ['CPUExecutionProvider']
```

However, OpenPilot sets some options for the `CPUExecutionProvider` https://github.com/commaai/openpilot/blob/master/selfdrive/modeld/runners/onnxmodel.py#L48-L51 that won't be set in the fallback mode.

### OS Version

Debian Trixie

### openpilot version or commit

deb79a9c443fa1fc1499e40a97757c7d8b6af735

mimi89999, 2024-03-11:
Maybe we should also install `onnxruntime-openvino` on intel GPUs

adeebshihadeh, 2024-08-09:
Still an issue?

closed 2024-10-05 by adeebshihadeh