commaai/openpilot #31711: Strange behavior of OpenPilot experimental mode in the simulator

mimi89999, 2024-03-05:
### Describe the bug

Hello,
When working on the MetaDrive integration, I notice that OpenPilot behaves very strangely in the simulator. It randomly attempts to cross the solid yellow lane and make a U turn on the road. I inspected the data sent drom the bridge with Cabana and everything seems fine, so I don't think that it is an issue with the simulation itself. I suspect that maybe missing inputs to the model might be causing it. The simulation doesn't have mapsd and other services.

### OS Version

Debian Trixie

### openpilot version or commit

deb79a9c443fa1fc1499e40a97757c7d8b6af735

OxygenLiu, 2024-03-08:
the experimental mode works only when `Driving Personality` is `Relaxed`.

adeebshihadeh, 2024-08-09:
Simulator is still pretty early, so it's not ready for bug reports like this yet. Join the Discord for further sim discussion.

closed 2024-08-09 by adeebshihadeh