commaai/openpilot #31713: [100$ bounty] ability to modify any part of a platformconfig with python

jnewb1, 2024-03-05:
we'd like the ability to modify the platform configs automatically using python, for automating spec updates, new model years, etc.

example:

```python
from selfdrive.car.honda.values import CAR

original_config = CAR.HONDA.config

new_config = original_config.copy()

new_config.specs.mass = 1400
new_config.flags |= HondaFlags.EXAMPLE_FLAG

save_config(CAR.HONDA, new_config)
```

and values.py reflects this change with no other diffs

webbdays, 2024-03-06:
I would like to work on this issue.
And also i would like to know more on this platform config.
where is this config file located.
Please provide any example use case.

jnewb1, 2024-03-06:
@webbdays `selfdrive/car/honda/values.py` is one example

teodoramilanovic, 2024-04-30:
#32327

teodoramilanovic, 2024-04-30:
#32327 - Could you check out this PR?

closed 2024-06-10 by sshane