commaai/openpilot #31714: Better simulator controls

mimi89999, 2024-03-05:
As discussed in https://github.com/commaai/openpilot/pull/31710 , the simulator controls are far from ideal. They are flickering and it is difficult to provide multiple inputs simultaneously.

To better capture keyboard events, we would need to open a window and that window would have to be focused. We could use Qt, Pygame (it's already a dependency of MetaDrive) or possibly another lib.

However just capturing keyboard events doesn't give much granularity. Maybe we could use sliders, measuring the pointer coordinates inside the box or some other method to better control that?

jnewb1, 2024-03-07:
I think a small pyqt ui for manual control of the simulator would be nice. we can also show the steering wheel angle, and a little mock dashboard that shows what the car would show (set speed, cruise status, etc)

MikeBusuttil, 2024-12-16:
does steering work for you guys?  My [A] and [D] keys appear to do nothing