commaai/openpilot #31745: Repeated LKAS errors since upgrading to 0.9.6

davetapley, 2024-03-06:
### Describe the bug

Obviously can't _guarantee_ causality, but have never seen this once until upgrading to 0.9.6, and has occurred 3 times now :scream:

Experimental off, nothing obviously _weird_ happening,
and then suddenly OP disengages, Comma 3X show "LKAS error",
dash shows ⚠️ and starts scrolling though:
1. AEB unavailable
2. Front camera unavailable
3. See dealer
4. Something else I can't remember but it's obviously very unhappy about the state of the car

I can drive manually just fine, and ignition on/off has 'fixed' it every time (so far 🤞🏻).

### Which car does this affect?

Toyota Highlander Hybrid 2019 with C-SDSU

### Provide a route where the issue occurs

fbdb5af0094cfbeb/2024-03-04--16-03-34/71

### openpilot version

0.9.6

sshane, 2024-03-07:
This looks to be a bug in the SDSU firmware, it started forwarding the message from the DSU in that segment you shared:

[image]

In 0.9.6 we shipped radar disable support for the new RAV4 so we check that no other ECU is sending the message on the bus, before this change openpilot would not have noticed the potential controls collision.

closed 2024-03-07 by sshane

adeebshihadeh, 2024-03-07:
I think it's premature for that. Let's just keep tracking it. Ideally we just have a software solution for them.

davetapley, 2024-03-07:
Is there anything I can do to help resolve?

@adeebshihadeh by 'software solution' do you mean something in OP to work around the SDSU bug, or a patch for the SDSU itself (which, I'm not even sure is patchable 😬)?

adeebshihadeh, 2024-03-07:
I mean disable the DSU in software like we do with the new Rav 4's. You'd still most likely lose AEB, but we plan on shipping our own.

davetapley, 2024-03-07:
@adeebshihadeh gotcha, and just to be sure: is this unrelated to https://github.com/commaai/openpilot/pull/31706?

adeebshihadeh, 2024-03-07:
I don't entirely get the question, but that kind of weird HW is what we're moving away from. See the discussion about that in the Toyota Discord channel.

davetapley, 2024-03-07:
[Followed up on Discord](https://discord.com/channels/469524606043160576/524327905937850394/1215384950346489906):

TLDR for others: yes probably same underlying issue.

> The check is introduced in 0.9.6 because some Toyota now requires the radar to be disabled (aka should not see `ACC_CONTROL` from the car), but for users that use SDSU, this isn't really needed.

I'm gonna downgrade to 0.9.5 until a resolution emerges.

Thanks for fast reply 🙏