commaai/openpilot #31775: Ability to connect to guest networks that use a captive portal

AreYouRlyScared, 2024-03-07:
Allow OP to connect to WiFi networks that have a captive portal.

TylerSharpe11, 2024-04-14:
> What's the use case for that? I guess that most users will simply insert a SIM card in their Comma devices or connect to the WiFi network from their phones.

Uber drivers could want a portal for customers.

Pious1020, 2024-09-14:
@notAreYouScared, is this still open?

notAreYouScared, 2024-09-14:
It would've been a nice to have so i could connect it to my works guest wifi and watch drives. But I have since sold my c3 due to the incompatibility with GMs new GlobalB can bus

aubsw, 2025-03-29:
@adeebshihadeh How feature-complete/robust do you want he captive portal rendering solution to be? There's obviously a tradeoff between completeness and performance. A couple options:

1. [QT Web Engine](https://doc.qt.io/qt-6/qtwebengine-overview.html). 
   Pros: Easiest to use right now. 
   Cons: Heavy because of chromium, conflicts with raylib UI rewrite?
2. [WebKitGTK](https://webkitgtk.org). 
   Pros: lighter than chromium, fully featured. Probably better for the post-QT UI. 
   Cons: Possibly more code than QT solution.

To compare/contrast the performance impact, here's an unverified assessment from chatgpt:

```
Memory usage (basic page render):
- WebKitGTK: ~60-100MB
- Chromium: ~300-500MB

Disk size:
- WebKitGTK + dependencies: ~30-40MB
- Chromium: ~150-200MB
```

There are other rendering options that very lightweight, but they generally don't support JavaScript (may be an issue these days).

adeebshihadeh, 2025-03-29:
So both of those are pretty massive for this one feature, especially now that we're moving to raylib.

I think one potential way forward for this now is to somehow do it through your phone using [connect](https://github.com/commaai/connect), but even that's not that obvious (do we setup a bluetooth connection for the initial phone <-> comma 3X?). Unless you've got a clever idea here, I think we can actually close this one.

aubsw, 2025-03-29:
The only idea I have here is that we could display QR code on the comma 3x with bluetooth info, and the connect app can scan that to establish phone <-> comma 3X.

But I think transfering the auth info for the captive portal system might be in issue if it's mediated this way. I think auth here is based on MAC address? Which we could totally spoof on the comma... But this just seems ugly.

adeebshihadeh, 2025-03-29:
Yeah... let's just close.

closed 2025-03-29 by adeebshihadeh