commaai/openpilot #31813: Add ability to login to WIFI that requires username

BBBmau, 2024-03-09:
I noticed that `username` is unused in `networkManager`. It would make sense to support the use of it for when users require both a username and password to connect to a network.

mimi89999, 2024-03-11:
What username? The configuration for WPA2 Enterprise?

BBBmau, 2024-03-11:
> What username? The configuration for WPA2 Enterprise?

Yep!

BBBmau, 2024-03-11:
@mimi89999 it can just be that if you go ahead and fillout the username field then it will be set to using `WPA2-Enterprise`. Similar to how connecting to a hidden network will automatically set it to `wifi/hidden` as true.

mimi89999, 2024-03-11:
Where do you have a WPA2 Enterprise network that you would like to connect to? The username isn't enough as there are multiple methods in WPA2 Enterprise.