commaai/openpilot #31822: Controls Lagging - 2023 Hyundai Santa Cruz

CruiseBrantley, 2024-03-11:
### Describe the bug

Roughly once per day will receive the error "Controls lagging, take control" (paraphrased). However will continue to work with error subsiding very quickly.

### Which car does this affect?

2023 Hyundai Santa Cruz

### Provide a route where the issue occurs

https://connect.comma.ai/099fdbcfdfbba538/1710109177530/1710109206715

### openpilot version

4b93a6417c1a8e6c02e2e081e163a0cd380cc4ba

### Additional info

This isn't specific to this commit, has been occurring for some time in my parent's vehicle. They've been running a fork with the issue but I switched them to master-ci yesterday and it's still occurring.

CruiseBrantley, 2024-03-28:
A couple more segments with the issue: 099fdbcfdfbba538/00000034--1dafec9b74/47
099fdbcfdfbba538/00000032--a0548890bf/45

CruiseBrantley, 2024-05-03:
As an update, I've replaced the usb-c cable with another shipped from China and it has no effect on the issue occurring. Next I may try another harness box I have but it will be a couple weeks.

adeebshihadeh, 2024-05-17:
Tracking in #32286

closed 2024-05-17 by adeebshihadeh