commaai/openpilot #31843: [100$ bounty] get platform_str from the enum name

jnewb1, 2024-03-12:
we currently have two separate names for each platform in openpilot, for example

`BRONCO_SPORT_MK1` (enum name)
`FORD BRONCO SPORT 1ST GEN'

this adds unnecessary complexity and an extra line to every platform config. this bounty is for removing the platform string, and instead having every PlatformConfig have the platform_str of it's enum name automatically. i.e.

```python
BRONCO_SPORT_MK1 = FordPlatformConfig(
  FordCarInfo("Ford Bronco Sport 2021-23"),
  CarSpecs(mass=1625, wheelbase=2.67, steerRatio=17.7),
)

assert (BRONCO_SPORT_MK1.platform_str == "BRONCO_SPORT_MK1")
```

guanarp, 2024-03-13:
Hey I'm looking at values.py

To be clear: This issue is changin for example
`# Toyota
  ALPHARD_TSS2 = ToyotaTSS2PlatformConfig(
    "TOYOTA ALPHARD 2020",
    [
      ToyotaCarInfo("Toyota Alphard 2019-20"),
      ToyotaCarInfo("Toyota Alphard Hybrid 2021"),
    ],
    CarSpecs(mass=4305. * CV.LB_TO_KG, wheelbase=3.0, steerRatio=14.2, tireStiffnessFactor=0.444),
  )`

to this:
`# Toyota
  ALPHARD_TSS2 = ToyotaTSS2PlatformConfig(
    [
      ToyotaCarInfo("Toyota Alphard 2019-20"),
      ToyotaCarInfo("Toyota Alphard Hybrid 2021"),
    ],
    CarSpecs(mass=4305. * CV.LB_TO_KG, wheelbase=3.0, steerRatio=14.2, tireStiffnessFactor=0.444),
  )`

am I missing something?

adeebshihadeh, 2024-03-13:
Exactly. The new string should then become "ALPHARD_TSS2"

jnewb1, 2024-03-13:
you might be able to get this to work by subclassing `auto` and using `_get_next_value_` within the enum, because this flow will be very similar to this example:

```
class MyEnum(AutoStrEnum):
    STATE_1 = auto()
    STATE_2 = auto()
  
from enum import Enum

class AutoStrEnum(str, Enum):
    """
    StrEnum where enum.auto() returns the field name.
    See https://docs.python.org/3.9/library/enum.html#using-automatic-values
    """
    @staticmethod
    def _generate_next_value_(name: str, start: int, count: int, last_values: list) -> str:
        return name
```

sshane, 2024-03-13:
I think you might need to look into the metaclass `EnumType` since we rely on the configs being hashable with the string.

thenhnn, 2024-03-15:
```python
BRONCO_SPORT_MK1 = auto(), FordPlatformConfig(
  FordCarInfo("Ford Bronco Sport 2021-23"),
  CarSpecs(mass=1625, wheelbase=2.67, steerRatio=17.7),
)
```

Is this syntax okay?

I can try to make it without auto(), but I have failed to do that in good way because I can't get key of member in `__new__`. Or I'm missing something?

alexsteinh, 2024-03-15:
It's possible, but you need to create a custom EnumType and exploit the auto member name generator for your own purposes.

jnewb1, 2024-03-15:
```
BRONCO_SPORT_MK1 = auto(), FordPlatformConfig(
  FordCarInfo("Ford Bronco Sport 2021-23"),
  CarSpecs(mass=1625, wheelbase=2.67, steerRatio=17.7),
)
```

can PlatformConfig subclass from auto and have the same effect?

thenhnn, 2024-03-15:
> > BRONCO_SPORT_MK1 = auto(), FordPlatformConfig(
> > FordCarInfo("Ford Bronco Sport 2021-23"),
> > CarSpecs(mass=1625, wheelbase=2.67, steerRatio=17.7),
> > )
> > Is this syntax okay?
> 
> can PlatformConfig subclass from auto and have the same effect?

I don't think so, because entire `auto` class is replaced.

guanarp, 2024-03-15:
It seems like I got a solution using _generate_next_value_ that @jnewb1 mentioned.
I added the test and is good for models where I delete the string, i.e, that are in this format:

```python
# Toyota
  ALPHARD_TSS2 = ToyotaTSS2PlatformConfig(
    [
      ToyotaCarInfo("Toyota Alphard 2019-20"),
      ToyotaCarInfo("Toyota Alphard Hybrid 2021"),
    ],
    CarSpecs(mass=4305. * CV.LB_TO_KG, wheelbase=3.0, steerRatio=14.2, tireStiffnessFactor=0.444),
  )
```
However, is the car/\<brand\>/values.py file generated automatically? I cannot find a way to do that. Or, should I delete the string name from all the constructors?.
I'll make a draft PR in a couple of hours.

jnewb1, 2024-03-15:
> However, is the car/<brand>/values.py file generated automatically?

nope, this is manully created. So yo have to delete it from all constructors manually

thenhnn, 2024-03-16:
> I'll make a draft PR in a couple of hours.

I'm already working on this, I will commit rewrite with EnumType soon.

thenhnn, 2024-03-16:
> > I'll make a draft PR in a couple of hours.
> 
> I'm already working on this, I will commit rewrite with EnumType soon.

I think I'm done, @jnewb1, can you approve CI?

closed 2024-03-19 by jnewb1