commaai/openpilot #31876: controlsd: adds steer saturated event prematurely

sshane, 2024-03-15:
The previous logic checked the actuators _output_, considering torque rate limits. We should continue to do the same.

@jnewb1 can you use process replay on a few segments with these events to make sure there's no (or expected minimal) diff after fixing? 

This core logic should not regress in refactors. I'm not sure how we want to go about testing for similar bugs in the future, but we should at least consider how things _might_ regress first.

jnewb1, 2024-03-15:
I like the idea of more process replay segments. We should try to hit 100% codecov for the core services. Perhaps we do process replay on fuzzed segments?

jnewb1, 2024-03-15:
(for example this part is never hit in any of our tests, even process replay
[image]