commaai/openpilot #31884: 2022 Kia Niro phev lxs fingerprint

BlueLightingeuv, 2024-03-15:
### Describe the bug

Comma 3 device states the 2022 Kia Niro Phev LXS version is unsupported, even though it is a supported vehicle.   I checked the Harness ordered from Comma Ai, and everything is connected. I appreciate any assistance that can be provided.

### Provide a route where the issue occurs

Dongle id=c7c920d630c75e71

### openpilot version

0.9.4

sshane, 2024-03-16:
Can you take a photo of your steering wheel, where the cruise control button is?

sshane, 2024-03-16:
Unfortunately only the EX and EX Premium have Smart Cruise Control, which is needed: https://www.kia.com/us/en/vehicles/niro-plug-in-hybrid/2022/specs-compare

[image]

---

Sorry about the inconvenience, I will update our car documentation for these cars. You can open a return here: https://shop.comma.ai/a/returns

closed 2024-03-16 by sshane