commaai/openpilot #31915: camerad: QGPUTargetObjGen.cpp:962: virtual bool llvm::QGPUModuleEncoder::runOnModule(llvm::Module&): Assertion

bccw2021, 2024-03-19:
### Describe the bug

camerad: QGPUTargetObjGen.cpp:962: virtual bool llvm::QGPUModuleEncoder::runOnModule(llvm::Module&): Assertion

I use my comma C3（comma.ai stock） in this way. Can you see where the problem lies?

1.
git clone --recurse-submodules https://github.com/commaai/openpilot.git
cd openpilot
git lfs pull
scons -u -j8

2.
The system prompted me to upgrade the AGNOS, and I agreed to install it

3.
comma@comma-f386829e:/data/openpilot$ ./system/camerad/camerad
camerad: /local/mnt/workspace/lnxbuild/project/trees_in_use/free_tree_dir/checkout/poky/build/tmp-glibc/work/sda845-oe-linux/adreno/1.0-r0/adreno200/llvm/lib/Target/Oxili/QGPUTargetObjGen.cpp:962: virtual bool llvm::QGPUModuleEncoder::runOnModule(llvm::Module&): Assertion 0 && "The total instruction number in CreateLabelPass does not match the total number after CodeGen"' failed. FATAL ERROR: Assert or abort fired (SIGABRT) common/clutil.cc: build failed; status=-2, log: Pass camerad: common/clutil.cc:91: cl_program cl_program_from_source(cl_context, cl_device_id, const std::string &, const char *): Assertion 0' failed.
Aborted (core dumped)

### Provide a route where the issue occurs

123456

### openpilot version

0.9.6

bccw2021, 2024-03-19:
I use https://flash.comma.ai/ flash the system also doesn't work！！！

adeebshihadeh, 2024-03-19:
"/local/mnt/workspace/lnxbuild/project/trees_in_use/free_tree_dir" is definitely not stock. Please do not spam our repo with anymore issues like this.

closed 2024-03-19 by adeebshihadeh