commaai/openpilot #31919: simulator is broken because of pedal removal

jnewb1, 2024-03-19:
simulator used pedal for long control. we should think of a way to separate the simulator from being tied to a specific car

jyoung8607, 2024-03-20:
It could be good to have an inverted CarState for each car port, call it SimCarState or whatever you like. Set up with a full config through get_params as normal, but instead of producing a CarState from CAN traffic, produce CAN traffic from a CarState.

Pros:
* Would enable platform-independent 100% test coverage of things like block engage on wrong gear, seatbelt, etc
* Would supersede and enhance the manual CAN packer tests in Panda, much easier to add new safety checks if desired
* Becomes almost free to drive any car in the simulator

Cons:
* Labor intensive to build for each car port, perhaps less with a thoughtful CarState refactor

jnewb1, 2024-03-20:
I like an inverted can state. I'd bet we could make it automatically with some mocking to see what each DBc field maps to

jyoung8607, 2024-03-20:
I like it more and more as I think about it. As soon as you have an inverted CarState for a port, you can start generating arbitrary car simulations in addition to the normal replay routes, and test them at different levels including full HITL. Can check race/timing issues that could cause controls mismatch, falsify checksums, fuzz counters, drop frames entirely, etc etc. Or if you woke up tomorrow and wanted Panda safety to verify seatbelt state, you could write it and validate it without leaving your desk.

mimi89999, 2024-03-23:
Hello,
I think that the reverse car state might be useful for unit tests, but I don't think that it's feasible to run a dozen simulations in the CI.

sshane, 2024-03-27:
The fuzzing stuff should be covered well by https://github.com/commaai/openpilot/blob/89f1483ac910c438719ca27fbff6b684e9922185/selfdrive/car/tests/test_models.py#L328

mimi89999, 2024-03-27:
I think that the best way to solve this would be to either simulate a different car that has unrestricted controls or add a "simulator" car. Not sure which variant is preferred by OP devs.

jnewb1, 2024-04-02:
we switched the car to one that can steer to zero so this issue is closed, but we can revisit the idea of an inverted car state

closed 2024-04-02 by jnewb1