commaai/openpilot #31970: Chrysler Pacifica Hybrid 2022 can not operate with multi warning lights when comma AI harness plugged in

gengyumichael, 2024-03-22:
### Describe the bug

Car model: 2022 Chrysler Pacifica Hybrid
Comma X3 with 0.9.6
First time plug in and power the car.
All cable seated properly and used other cable to test as well.
Car Behavior:
multi warning lights on, can not start the car. Comma can not recognize the car

### Which car does this affect?

Chrysler Pacifica 2022

### Provide a route where the issue occurs

b7b92352c118d69f/2024-03-21--21-22-52/0

### openpilot version

0.9.6

gengyumichael, 2024-03-22:
The FCA harness somehow when it is plugged non. Car is doing crazy things like all lights on, keeping beeping. If I plug the default camera harness in. It works fine

sshane, 2024-04-15:
Does this still happen?

sshane, 2024-04-15:
Closing since you returned your device it looks like

closed 2024-04-15 by sshane

gengyumichael, 2024-04-15:
this got resolved with a new harness BestYu Geng

Closed #31970 as completed.