commaai/openpilot #31987: Mazda 2024 CX5 Carbon Dashcam Model, Car Unrecognized

wenqian-ye, 2024-03-23:
### Describe the bug

See https://github.com/commaai/openpilot/issues/30872#issuecomment-2008482381.

dongle id: 833d1eb9ecc9b842

### Which car does this affect?

Mazda 2024 CX5 Carbon

### Provide a route where the issue occurs

833d1eb9ecc9b842/2024-03-23--05-59-28/3

### openpilot version

0.9.6-release

### Additional info

This issue happens on all the recorded drives. I have double checked the connection of cables.

sshane, 2024-03-25:
Your car should be supported on master

closed 2024-03-25 by sshane

wenqian-ye, 2024-03-25:
Could you provide the url corresponding to the branches? I was using the default openpilot.comma.ai

sshane, 2024-03-25:
openpilot-nightly.comma.ai

wenqian-ye, 2024-03-25:
Thanks! I will take a try and let you know how it works.

wenqian-ye, 2024-03-26:
It works on my car with the master branch. Thank you again!