commaai/openpilot #32005: “Process not running:qcomgpsd”

haram-KONA, 2024-03-26:
### Describe the bug

“Process not running:qcomgpsd” that repeats every day.
When I first drive every morning, "Process not running: qcomgpsd" occurs, and if I "reboot" the C3X while driving in this state, it usually returns to normal. 
I wonder why this phenomenon repeats itself. 
Is it a hardware problem or a software problem? 
I have also attached a photo of the condition. 
photo "1" is always checked when driving for the first time because "qcomgpsd" is suspicious, and photo "2" is a state that occurs when i engage OpenPilot in the state of "1". 
If i "reboot" in "2" status, it almost returns to normal.

route : 1618132d68afc876/0000001b--4188a680db/0
        1618132d68afc876/0000001b--4188a680db/1
        1618132d68afc876/0000001b--4188a680db/2

"notice"
In this situation, the route is registered in “manage account”, but not in “connect”. 

Please refer to the information below.
 
Currently, I am a user living in South Korea. 
In the past, I was subscribed to Comma Prime, but according to Comma's policy, I am currently subscribed to Comma Prime Lite. 
In the past, under Comma Prime, the problem i asked about never occurred, and after changing to Comma Prime Lite, LTE Internet use using a SIM card was restricted, so I had no choice but to use the APN Setting of KT (lte.ktfwing.com), an internet service provider in Korea, where I live. 
I was using the Internet, and this problem seems to have occurred while using the APN Setting for some time. 
About two days ago, I connected to the Internet using my cell phone hotspot with only the SIM card inserted without setting the APN settings, and the problem I mentioned does not occur. 
So, my guess is that a problem occurs when using Korea Telecom's APN Setting (lte.ktfwing.com). 
Currently, this problem does not occur with a cell phone hotspot without APN Setting. 
Please check.
[image]
[image]

### Provide a route where the issue occurs

1618132d68afc876/0000001b--4188a680db/0         1618132d68afc876/0000001b--4188a680db/1         1618132d68afc876/0000001b--4188a680db/2

### openpilot version

0.9.7 (master-ci) & 0.9.6 (release 3)

### Additional info

this route that problem does not occur with a cell phone hotspot without APN Setting. 

1618132d68afc876/0000000d--bd37ebda6a

adeebshihadeh, 2024-03-26:
So it definitely seems related to your 3rd party SIM, but it's not clear how. I don't see anything that indicates this is a HW issue.

Can you try this branch? `installer.comma.ai/commaai/noxtratime`

haram-KONA, 2024-03-26:
> So it definitely seems related to your 3rd party SIM, but it's not clear how. I don't see anything that indicates this is a HW issue. The module is just not accepting the time assistance.
> 
> Can you try this branch? `installer.comma.ai/commaai/noxtratime`

Oh !! Ah~~ 
I installed and drove the branch you mentioned and the error did not occur. 
With LTE APN Setting enabled, LTE connection works well and GPS works well.
route id : 1618132d68afc876/00000000--4b43f08fa6/0

However, the route is saved in “manage account” "useradmin", but the route is not visible in connect.
The saved time and actual KOR Time do not match.

haram-KONA, 2024-04-15:
Is there no solution to this problem?

xxyk, 2024-09-04:
same problem. Do you find the solution ？

closed 2024-12-14 by adeebshihadeh