commaai/openpilot #32016: navd: fails to fetch route on first boot

sshane, 2024-03-26:
This is caused because the time is usually not correct for the first ~10 seconds after boot/onroad, and when we get a token from the API we provide it a date from November of last year:

https://github.com/commaai/openpilot/blob/d05c19b9f4b70b3ef1c5c39734cf5175c8c5c020/common/api/__init__.py#L25-L26

Which causes this exception in navd until navd is restarted:

```
{"event": "API request failed", "status_code": 403, "text": "<!doctype html>\n<html lang=en>\n<title>403 Forbidden</title>\n<h1>Forbidden</h1>\n<p>Error getting device info with provided JWT</p>\n", "error": true}
```

adeebshihadeh, 2024-03-26:
Got a route? `chronyd` should pull the time pretty quickly once you have internet.

sshane, 2024-03-26:
e9b1885ec511d838/00000029--74ccd3fd9c/14

PR #32017 (2024-03-27): navd: generate new JWT for each request
fixes #32016
closed 2024-03-27 by adeebshihadeh