commaai/openpilot #32023: 2017 Chevy Volt ACC Cruise Fault

Ohiovolt, 2024-03-27:
### Describe the bug

I'm using C3x with L&P harness and latest release of OP. L&P harness is functioning properly. C3x device states Cruise Fault: restart car to engage. Turning off car does not resolve issue. I have tried rotating the USB-C cable and it did not fix the issue. I have not been able to get this to work since installing the first time. 
[image]
[image]

### Which car does this affect?

2017 Chevy volt with ACC

### Provide a route where the issue occurs

8c4958bc2fe68698/2024-03-26--22-57-18/0

### openpilot version

0.9.6

sshane, 2024-03-27:
Can you try to set `SKIP_FW_QUERY=1` before launching openpilot and try again?

Such as: Kill openpilot and run `SKIP_FW_QUERY=1 /data/continue.sh`

Ohiovolt, 2024-03-27:
Yes. It has acc. I'm not sure how to do that. Would it be in settings?

sshane, 2024-03-27:
You will want to ssh into the device: https://github.com/commaai/openpilot/wiki/SSH

Ohiovolt, 2024-03-27:
I am able to ssh in. Can you please share the command prompts I should type in to achieve that result

sshane, 2024-03-27:
You can place this line at the top of /data/continue.sh

`export SKIP_FW_QUERY=1`

Ohiovolt, 2024-03-27:
I really do apologize and I don't mean to sound rude but I'm not a developer and will need a little more guidance. I don't even know how to get to /data/continue.sh to include that line of code

Ohiovolt, 2024-03-27:
I ran the following command prompts:

echo -en "1" > /data/params/d/DisableUpdates
Then
echo "export SKIP_FW_QUERY=1" >> /data/openpilot/launch_env.sh

I then rebooted the device and went for a drive. It did not resolve the issue.

sshane, 2024-03-28:
That did not seem to take. Try this command once, then reboot the device and start the car once the device is started again: `sed -i '1 aexport SKIP_FW_QUERY=1' launch_openpilot.sh`

Ohiovolt, 2024-03-28:
I followed your instructions but not sure this one took either as I'm still experiencing the same issue. 

Route: 8c4958bc2fe68698/2024-03-28--09-19-10/0

Ohiovolt, 2024-03-28:
Do you have any other suggestions? Do you think it's a hardware or software issue? Should I try installing a previous version of OP or just return the c3x for a new one?

sshane, 2024-03-29:
After 10 seconds or so, boardd thinks that sendcan is more than a second off and it stops sending the CAN messages, but in the logs the timing looks all correct for sendcan and can. @adeebshihadeh do you have any ideas?

[image]

The first time is `nanos_since_boot()` in boardd and the rate they both increase don't match.
```
[24.099975] selfdrive/boardd/boardd.cc:232 - can_send_thread: sendcan too old to send: 251973676638, 250688230649
[24.100018] selfdrive/boardd/boardd.cc:232 - can_send_thread: sendcan too old to send: 251973676638, 250688230649
[24.100253] selfdrive/boardd/boardd.cc:232 - can_send_thread: sendcan too old to send: 251973707003, 250699614972
[24.100209] selfdrive/boardd/boardd.cc:232 - can_send_thread: sendcan too old to send: 251973707003, 250699614972
[24.100467] selfdrive/boardd/boardd.cc:232 - can_send_thread: sendcan too old to send: 251973762419, 250709298513
[24.100425] selfdrive/boardd/boardd.cc:232 - can_send_thread: sendcan too old to send: 251973762419, 250709298513
[24.100677] selfdrive/boardd/boardd.cc:232 - can_send_thread: sendcan too old to send: 251973793096, 250721121013
[24.100635] selfdrive/boardd/boardd.cc:232 - can_send_thread: sendcan too old to send: 251973793096, 250721121013
[24.100882] selfdrive/boardd/boardd.cc:232 - can_send_thread: sendcan too old to send: 251973839659, 250728695545
```

clementcho, 2024-03-30:
> You can place this line at the top of /data/continue.sh
> 
> `export SKIP_FW_QUERY=1`

Should I put the line before or after the #!/usr/bin/bash

Thanks,
Clement

clementcho, 2024-03-30:
Okay, so I ran a Fingerprinting 2.0 procedure from here: https://github.com/commaai/openpilot/wiki/Fingerprinting 
I got carFw = [ ]. That would definitely cause a problem. My route is 9dad298fbcfbca02%7C00000004--1a720dd784
I also attached the segment 0 txt in this. 

[image]
[Route_9dad298fbcfbca0200000004_1a720dd784.txt](https://github.com/commaai/openpilot/files/14813029/Route_9dad298fbcfbca0200000004_1a720dd784.txt)

sshane, 2024-04-01:
Can you try the master branch? I can give you a branch to debug this after

Ohiovolt, 2024-04-02:
How would I do this?

sshane, 2024-04-02:
Use this installer URL: https://installer.comma.ai/commaai/master-ci

Ohiovolt, 2024-04-02:
Ok it is installed but it is giving me the same cruise fault.

sshane, 2024-04-02:
Okay, now try this one: https://installer.comma.ai/commaai/volt-no-fault

Ohiovolt, 2024-04-02:
Do I need to uninstall this first?

Ohiovolt, 2024-04-02:
It said "failed to check for update"

Ohiovolt, 2024-04-02:
Still getting cruise fault

Ohiovolt, 2024-04-02:
I need to know if this is a software or hardware problem and if this is compatible with my vehicle or if I need to return this.

adeebshihadeh, 2024-04-02:
I strongly suspect this is a software issue. Feel free to return it now, but we'll extend your return window if you're willing to wait a bit.

@sshane let's get this resolved ASAP.

Ohiovolt, 2024-04-02:
Yes absolutely. Thank you. I'm willing to work with your team to get this resolved. I literally just purchased this car 3 weeks ago for the sole purpose of being able to use OP.

sshane, 2024-04-08:
@Ohiovolt can you pull the branch I gave you and retry? I added extra logging.

Ohiovolt, 2024-04-08:
Sorry just to confirm, By pull do you mean uninstall and reinstall?

sshane, 2024-04-08:
You can check for updates in the UI.

Ohiovolt, 2024-04-08:
@sshane it seems to be stuck on finalizing update. It's been on this for over 20 min. [image]

Ohiovolt, 2024-04-08:
This is the error I'm getting now. [image]

sshane, 2024-04-08:
we're checking it out

sshane, 2024-04-08:
Can you uninstall and re-install? We're currently re-working the updater

Ohiovolt, 2024-04-08:
@sshane latest route using updated branch 

https://connect.comma.ai/8c4958bc2fe68698/1712616517731/1712616795935

sshane, 2024-04-08:
It didn't upload the first rlog, potentially because the log file was too big. Can you start the car for a max of 15 seconds or until it faults and then turn it off?

Ohiovolt, 2024-04-08:
Try this: 

https://connect.comma.ai/8c4958bc2fe68698/1712616828393/1712616847031

sshane, 2024-04-09:
I pushed an update to the branch, does fetch updates work? If not, you can ssh in and `git pull` and reboot (fastest), or re-install (slowest).

sshane, 2024-04-09:
I think this may be a potential hardware failure with your harness or the panda, unsure yet. We are not seeing any CAN traffic on bus 2/chassis bus and that CAN core is returning many stuff errors. Since we send a friction brake command on that bus, the tx buffer is likely filling up and causing the panda to return `NACK` as it's not ready for more messages yet.

@Ohiovolt Can you describe your hardware set up? Which harness did you get for the ASCM? There's a few options you could have done here: https://github.com/commaai/openpilot/wiki/GM#hardware

Ohiovolt, 2024-04-09:
I think I pulled the update but now it doesn't seem to be syncing with connect.

Ohiovolt, 2024-04-09:
https://connect.comma.ai/8c4958bc2fe68698/1712623924067/1712623945684

Ohiovolt, 2024-04-09:
I am using the L&P harness with c3x and followed the instructions as detailed. ACC works when c3x is not plugged in.

Ohiovolt, 2024-04-09:
I did receive a canbus error before I reboot after the pull and tried flipping the usb-c around. This is the most recent route. 

https://connect.comma.ai/8c4958bc2fe68698/1712624150640/1712624165661

sshane, 2024-04-09:
Have you tried a new USB C cable yet? The errors switch sides when you flipped the cable.

Ohiovolt, 2024-04-09:
I can try using the other shorter usb-c cable that came with the hardware.

Ohiovolt, 2024-04-09:
Using the shorter cord that came with the hardware with the 90 degree seemed to have given the car more problems:
https://connect.comma.ai/8c4958bc2fe68698/1712626264948/1712626283746

USB Flipped:
https://connect.comma.ai/8c4958bc2fe68698/1712626356892/1712626391018

sshane, 2024-04-09:
It looks like the culprit is either the comma OBD connector or the third party L&P harness. Because it is difficult to diagnose here, I suggest you consider making a return request at https://shop.comma.ai/a/returns

Ohiovolt, 2024-04-09:
Why did you close this issue? Are you able to send me a replacement device in the meantime?

clementcho, 2024-04-09:
Sshane, may I ask what's is the new comma power need to read from CANS that is not getting on Gray Panda or the old Comma Power on Comma 3? Or if the Comma 3 and 3X are using the same Comma Power, then the culprit would fall into the Red Panda (I assume the built-in panda in comma 3x is red). Given I can see Comma 3 was able to be successfully installed on Volt 2017 as the YouTube demo shows here https://youtu.be/QeMCN_4TFfQ I got this link from the openpilot code.

Ted2645, 2024-04-10:
Hi, im having this exact same issue. I just got.my l and p harness and have been going through everything on the install triple checking my cabling thinking I screwed something up. Is there a fix? I litterally bought this specific volt to use with comma ai. I am willing to work with anyone to figure this out! I hope ohiovolt didn't give up. 

https://github.com/commaai/openpilot/assets/145208556/7b74e499-0b74-4a40-a34c-9a26b63fa88a

https://github.com/commaai/openpilot/assets/145208556/10e2d7e8-77e4-4480-981e-694ba03043bf

Ted2645, 2024-04-10:
Hey! Hopefully Ohiovolt didn't take the comma back I fixed mine.... it was the switch at the back near the ascm I'll attach a picture! Make sure it's switched to "op"

https://github.com/commaai/openpilot/assets/145208556/64ad2e5f-ff7e-4798-8a11-40cdd0b8ecd6

[image]

Ted2645, 2024-04-10:
Mine seemed like it was set to op, but I redid it and bam calibrating! Great success. Can't wait to try out experimental

Ohiovolt, 2024-04-10:
Unfortunately I believe the issue is with my comma obd adapter. I am hoping they send me hardware. Still waiting on their response from Sam. Sent from my iPhone

https://github.com/commaai/openpilot/assets/145208556/64ad2e5f-ff7e-4798-8a11-40cdd0b8ecd6
20240410_175506.jpg (view on web)

Ted2645, 2024-04-10:
They told me I would need to buy another obd2 usbc sensor from them, they wouldn't send me one for the trouble.  Said I would have 30 days from purchase to refund..... hope you get it worked out

clementcho, 2024-04-11:
> They told me I would need to buy another obd2 usbc sensor from them, they wouldn't send me one for the trouble. Said I would have 30 days from purchase to refund..... hope you get it worked out

Did you say you are actually able to drive your Volt with Comma 3X after you switch from OP to Dealer mode? Have you done the road test yet?

Ted2645, 2024-04-11:
Yes, I got it working

Ted2645, 2024-04-11:
Funnily enough I installed sunnypilot and messed around with cabling a bit more (mainly that switch in the back) and it started working after 4 hours of troubleshooting. I would run through all tour cabling again. I took this car apart 4 times :(

Ted2645, 2024-04-11:
Also check this fuse, I had mine pointed up and if was causing issues, 

[image]

clementcho, 2024-04-12:
Actually Dealer mode means the cabling won't bypass Ascm. That would means comma 3X would be no longer need L&P harness. Are you sure it is really what you did? Just flip that dealer switch?

berriosa13, 2024-09-03:
I am having the exact same issue with my 17' premier w/ ACC. Using L&P harness as well as ODB2 harness. Have switched between dealer/OP mode, and I tried different type c cables, all to no avail. Thinking it may be the OND2 harness at this point, due to the fact them when toggling the stock ACC to OP switch in the front of the car, ACC stock works fine. As soon as I swap to OP, ACC fault code goes off and comma displays cruise fault error.

berriosa13, 2024-09-03:
@Ted2645, as @clementcho stated, ASCM doesn't get bypassed when dealer mode is set, so did you switch yours from OP to dealer and then back to OP again, or is your vehicle someone working while dealer mode is switched on?

Ted2645, 2024-09-03:
I donno I swapped that switch from op to dealer back to op, and installed sunypilot and it started working. It does randomly "fault out" but I just restart the comma and it works again.....

berriosa13, 2024-09-03:
@Ohiovolt Did you ever end up figuring this out? I have the same make and model volt and I'm having the same exact issue. Tried nearly everything you have and installed countless forks.