commaai/openpilot #32027: Error: Communication Issue Between Processes deviceState, managerState

wowLookAmazingUsername, 2024-03-27:
### Describe the bug

Communication issues between processes

**Describe the bug**
Running into this error that disables OP shortly after starting up: Communication Issue Between Processes deviceState, managerState

**Which car does this affect?**
Toyota RAV4 2021

**Provide a route where the issue occurs**
4baf26681af2f7bd/2024-03-26--07-53-12/0

**openpilot version**
0.9.6
[image]

### Provide a route where the issue occurs

4baf26681af2f7bd/2024-03-26--07-53-12/0

### openpilot version

0.9.6

### Additional info

Been experiencing this across different versions of OP for the past maybe 6-8 months

wowLookAmazingUsername, 2024-03-27:
Did it again today:
4baf26681af2f7bd/2024-03-27--07-40-56/0

adeebshihadeh, 2024-03-31:
Duplicate of #32034

closed 2024-03-31 by adeebshihadeh