commaai/openpilot #32034: Low memory crashing OP

wowLookAmazingUsername, 2024-03-28:
### Describe the bug

OP runs out of memory, suspecting a memory leak, leading memory to fill up and eventually crashing OP within minutes.
[image]

### Provide a route where the issue occurs

4baf26681af2f7bd/2024-03-27--21-27-36/0

### openpilot version

0.9.6

### Additional info

During the same route, the error “low process communication rate” also showed up twice

wowLookAmazingUsername, 2024-03-28:
> Something is going on with the UFS flash on your device. I've got a device here on my desk with similar symptoms that I recalled recently. I'll look into this soon and hopefully it'll be a software fix.

Is that the reason I’m getting the communication issues between processes error too?

adeebshihadeh, 2024-03-28:
Yes

wowLookAmazingUsername, 2024-04-11:
Updates?

wowLookAmazingUsername, 2024-04-11:
Is there any updates? @adeebshihadeh

ThomasBurgess2000, 2024-06-30:
Got this issue today, restarted and it is persisting https://github.com/commaai/openpilot/issues/32027

Sliicy, 2024-11-21:
I have the same issue ever since I got a replacement Comma 3X; every so often this happens. Since I'm running SunnyPilot, I will need to move back to Stock to see if this happens, and then link a route for the devs to see.

pedrosimao, 2025-07-19:
Hey,
I just bought a Comma 3X and on the second drive after installing I started getting these `communication issue between processes deviceState` errors. Quite frustrating. Should I send my device for a replacement?
I live in the EU and I already paid like 30% taxes over the price. Hope I won't need to replace the device.

ajlunis, 2025-10-18:
Any updates on this? I seem to be experiencing the same issue after 14 months in use with no issues or software changes @adeebshihadeh