commaai/openpilot #32035: LogReader doesn't work with non-continguous segments

adeebshihadeh, 2024-03-28:
Needs a better error and some way to just get whatever segments are present

```
batman:debug$ ./count_events.py "4baf26681af2f7bd/2024-03-27--21-27-36"
Traceback (most recent call last):
  File "/home/batman/openpilot/selfdrive/debug/./count_events.py", line 23, in <module>
    for msg in LogReader(sys.argv[1], ReadMode.QLOG):
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/batman/openpilot/openpilot/tools/lib/logreader.py", line 256, in __init__
    self.reset()
  File "/home/batman/openpilot/openpilot/tools/lib/logreader.py", line 279, in reset
    self.logreader_identifiers = self._parse_identifiers(self.identifier)
                                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/batman/openpilot/openpilot/tools/lib/logreader.py", line 239, in _parse_identifiers
    identifiers = source(sr, mode)
                  ^^^^^^^^^^^^^^^^
  File "/home/batman/openpilot/openpilot/tools/lib/logreader.py", line 191, in auto_source
    raise Exception(f"auto_source could not find any valid source, exceptions for sources: {exceptions}")
Exception: auto_source could not find any valid source, exceptions for sources: [AssertionError(), AssertionError(), AssertionError(), AssertionError()]
```

sshane, 2024-03-28:
FYI error message was more clear here: https://github.com/commaai/openpilot/pull/31484

sshane, 2025-07-18:
Your example is already fixed

closed 2025-07-18 by sshane