commaai/openpilot #32066: Getting car unrecognized message on supported car

mrkushaljoshi, 2024-04-01:
### Describe the bug

Comma3X displays "Dashcam Mode car unrecognized" intermittently. I would say around 50% of the time I turn on my vehicle this occurs. Alongside that when I remove the USB and plug the unit back in it syncs up fine. 

Comma3X is hooked up to a fully supported car: Honda CRV 2019 EX

### Which car does this affect?

Honda CRV 2019 EX

### Provide a route where the issue occurs

4a76a1c956f48b333|2024-03-22-23-00-48

### openpilot version

0.9.6

sshane, 2024-04-01:
Can you plug the device in to constant power, or use your comma power? I don't see any of the non-fingerprinting routes uploaded at all. The only two are on March 19 an d22

mrkushaljoshi, 2024-04-01:
@sshane I have only been utilizing comma power with my device. Do I need to just keep it connected to my hotspot for a while for it to upload correctly? Thanks in advance.

mrkushaljoshi, 2024-04-01:
@sshane connected to another power source and let it upload. Validated on Comma Connect that Upload queue now shows
no uploads Please let me know if you need me to take anymore action on my end! Cheers.

sshane, 2024-04-08:
That's CAN fingerprint issue is unrelated.

sshane, 2024-04-08:
I'll get to checking this out soon, currently busy with another project.

sshane, 2024-04-15:
@mrkushaljoshi can you check out master and try again?

https://installer.comma.ai/commaai/master

mrkushaljoshi, 2024-04-30:
@sshane,

Went ahead and reset + reinstalled openpilot via https://installer.comma.ai/commaai/master. Issue persists about the same amount as before and now occasionally get 'Branch not tested' error on boot. 

Plugging it back in when I get the error resolves is around 3/4 of the time, if I plug it in a second time I have had 100% success rate so far. 

Let me know if there is anything else I can try! 

- @mrkushaljoshi

sshane, 2024-05-08:
You've only tried release3, you need to try the branch link above and connect to WiFi so the logs can upload.

mrkushaljoshi, 2024-05-20:
@sshane,

Forgot to ping back here but the Comma3x was connected to Wi-Fi and logs updated! Let me know if you need anything else from my end! 

- @mrkushaljoshi

sshane, 2024-05-31:
Requested full log upload of a recent drive, please put it on WiFi again

haram-KONA, 2024-05-31:
> Requested full log upload of a recent drive, please put it on WiFi again

It seems to rarely appear these days. I think May 22nd was the last day it occurred. 
The route is "1618132d68afc876/0000001d--ade82d6865/0", but the log is not saved well.

These days, I almost give up and just turn it off and on when it happens and drive, but it doesn't seem to happen as frequently as it used to.

sshane, 2024-05-31:
@mrkushaljoshi do you have any additional diagnostic devices in the car?