commaai/openpilot #32070: “Dashcam Mode car unrecognized” occur.

haram-KONA, 2024-04-01:
### Describe the bug

When turn on the car while the comma C3X is connected to the car, “Dashcam Mode car unrecognized” will almost always occur. 
reboot the C3X in this state, it will engage normally again. 
After engaging, drive in openpilot normally and turn off the engine with the C3X connected and then turn it on again, the "Dashcam Mode car unrecognized" status will also occur, and reboot again, the engagement will return to normal. 
always turn on the ignition without connecting the USB-C of the C3X and then connect the USB-C cable, this problem will not occur.

### Which car does this affect?

Hyundai KONA EV 2ND GEN (HDA2)

### Provide a route where the issue occurs

1618132d68afc876/0000000f--3a20636e0f

### openpilot version

0.9.7 master-ci

mrkushaljoshi, 2024-04-01:
I have a similar issue #32066. Vehicle Make and Model different however but the behavior is oddly similar.

sshane, 2024-04-01:
Is this a new issue?

haram-KONA, 2024-04-01:
> @haram-KONA is this new?

It keeps repeating little by little, but these days, the cycle occurs frequently.

haram-KONA, 2024-04-02:
1618132d68afc876/00000011--e1f95ed2b7/0

This route occurred again this morning (KOR Time: Tuesday Apr 2 @ 09:45 - 09:45).
“reboot” in this state, Engage will operate normally again.
This happens over and over again, and the Trainning  value of the torque-control is reset each time start driving, so it training again every time i start driving.

haram-KONA, 2024-04-08:
Please save me~~~
The "mock" state keeps appearing and not appearing again and again, and the TorqueControl Training data disappears every day... 
so My Car just keep Training over and over again....

Please save me~~~

route : 1618132d68afc876/0000002a--865a3d5ef4

Verylukyguy, 2024-04-08:
I have been having a similar unresolved issue https://github.com/commaai/openpilot/issues/30404
Every other start up results in Dashcam Only mode and I need to restart the vehicle.

sshane, 2024-04-15:
@haram-KONA please try that branch and report back.

haram-KONA, 2024-04-16:
It is currently being used as " master-ci ", but it has not occurred yet since April 10th (KOR Time).

sshane, 2024-04-17:
There may be a few bugs in Hyundai's (VIN/) queries that were masked before this cleanup PR: https://github.com/commaai/openpilot/pull/32022

The cluster ECU we use to get the VIN on HDA1 cars for example can see our bus 0 queries on HDA2 cars, but is filtered out unless the multiplexing mode is set to the radar bus, not OBD port. Found by querying it on bus 0, then switching off OBD multiplexing: you see it on bus 1 and we don't response to it.

sshane, 2024-05-31:
@haram-KONA I requested upload of some recent routes, can you put it on wifi to upload?

haram-KONA, 2024-05-31:
> @haram-KONA I requested upload of some recent routes, can you put it on wifi to upload?

now connected wifi

haram-KONA, 2024-05-31:
> @haram-KONA I requested upload of some recent routes, can you put it on wifi to upload?

" Hyundai CAN FD: match cluster speed #30213 "  : Is this difficult to apply?

sshane, 2024-05-31:
Please try to reproduce it by rebooting the device a few times, then you can request upload those logs

haram-KONA, 2024-05-31:
> Please try to reproduce it by rebooting the device a few times, then you can request upload those logs

all logs uploaded

sshane, 2024-06-01:
So far we've discovered three things: on startup an unknown ECU requests DTC info from the camera (could be others as well):

```
    ( address = 1988,
      busTime = 0,
      dat = "\003\031\002\fUUUU",
      src = 1 ),
    ( address = 826,
--
    ( address = 1996,
      busTime = 0,
      dat = "\003Y\002\f\252\252\252\252",
      src = 1 ),
    ( address = 568,
```

and a few seconds later, it then requests the part number (0xF187). But this doesn't seem to prevent us from fingerprinting in most cases, as this is a bit after we should have fingerprinted.

The second discovery is that on this HDA2 car, you can query an ECU normally on non-OBD bus 1 on bus 0 and it will send you a response on bus 1 that we ignore because we're specifically looking for a response on bus 0. This could be putting the camera into a bad state as it waits for our flow control continue message that we never send.

The third discovery is sometimes our sent query doesn't return on bus 128/129 which could mean the camera never receives our message at all. I don't see any lost message increase in the few routes I've checked, so this could be an intermittent panda bug, possibly related switching to a disconnected comma power that racks up rx errors.

sshane, 2024-06-01:
@adeebshihadeh I do actually see a dropped CAN message right at the exact timestamp that we're missing a response, do you want to check this out? `1618132d68afc876/00000020--f400d529c7/0`

[image]

sshane, 2024-06-01:
There's also some weird interaction with the OBD bus, it reports a response without a comma power! But no response if we don't query non-OBD right after, so there's some bug here. Will look into it

```
sendcan: rxaddr=1988, bus=1, 33983.36 ms, 0x0322f10000000000, b'\x03"\xf1\x00\x00\x00\x00\x00', len(can.dat)=8                                                                                           

    can: rxaddr=1988, bus=129, 33989.22 ms, 0x0322f10000000000, b'\x03"\xf1\x00\x00\x00\x00\x00', len(can.dat)=8

    can: rxaddr=1996, bus=1, 34129.61 ms, 0x103362f100535832, b'\x103b\xf1\x00SX2', len(can.dat)=8                                                                                                       
                                                                                                                                                                                                         
sendcan: rxaddr=1988, bus=1, 34201.5 ms, 0x0322f10000000000, b'\x03"\xf1\x00\x00\x00\x00\x00', len(can.dat)=8                                                                                            
                                                                                                                                                                                                         
    can: rxaddr=1988, bus=129, 34210.71 ms, 0x0322f10000000000, b'\x03"\xf1\x00\x00\x00\x00\x00', len(can.dat)=8
```

closed 2024-06-05 by sshane