commaai/openpilot #32129: Car Unrecognized on Mazda CX5 2024

Ariiees, 2024-04-08:
### Describe the bug

We connected the Comma 3 with Mazda CX5 2024, but get the "car unrecognized"  message. 
The Dongle ID is dcb2ee34f8c4c2be, the serial is 5185fe41. The current software version is 0.9.6-release/release3/e5d2ce5/Feb 27.
The Car is Mazda CX5 2024.

### Which car does this affect?

Mazda CX5 2024

### Provide a route where the issue occurs

None

### openpilot version

0.9.6-release

sshane, 2024-04-08:
Can you connect your device to WiFi for 30 minutes to upload logs? I don't see anything from the last few months. You can also try to re-install openpilot to start fresh.

Ariiees, 2024-04-23:
Hi sshane, 
Thank you so much for your help! We just re-install openpilot and test again. It still unrecognized. And the route number is dcb2ee34f8c4c2be/2024-04-23--18-16-52/0

Ariiees, 2024-04-24:
Hi sshane,

Thank you so much. It works now！