commaai/openpilot #32143: "Unrecognized car" on 2019 Honda Odyssey EX-L

djwhitt, 2024-04-09:
### Describe the bug

I replaced my existing working Comma 2 with a new Comma 3x and am now getting "Unrecognized car".

### Which car does this affect?

2019 Honda Odyssey EX-L

### Provide a route where the issue occurs

edf22f348d55c611/2024-04-09--14-02-26

### openpilot version

0.9.6

### Additional info

### comma 3X

|dongle_id|serial|device_type|alias|sim_id|prime|prime_type|last_ping||
|---|---|---|---|---|---|---|---|---|
|[edf22f348d55c611](https://useradmin.comma.ai/?onebox=edf22f348d55c611)|f1916592|threex|None|89852351123040019815|False|0|2024-04-09 14:03:52|uploads accepted|

### comma 2

|dongle_id|serial|device_type|alias|sim_id|prime|prime_type|last_ping||
|---|---|---|---|---|---|---|---|---|
[e026c3b5d55633ad](https://useradmin.comma.ai/?onebox=e026c3b5d55633ad)|b02cb0c4|two|None|8901260195759121761|False|0|2024-04-06 17:48:33|uploads accepted|

sshane, 2024-04-09:
We enabled fingerprinting without the comma power on master branch, which will come to release soon. Please check out master or nightly to use openpilot in the mean time?

https://installer.comma.ai/commaai/nightly

closed 2024-04-09 by sshane

djwhitt, 2024-04-09:
Wow, fast response. Thanks! I actually just came back here to say this was user error. I just forgot to reconnect my cat5 cable. :sweat_smile: