commaai/openpilot #32179: Toyota RAV4 XLE hybrid 2024 error my vehicle is not recognized.

yahaplus, 2024-04-12:
### Describe the bug

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

Good afternoon. Toyota RAV4 XLE hybrid 2024 (VIN: JTME6RFV6RD557044) error my vehicle is not recognized.
Received the Comma 3X yesterday. I set it up and went to connect it to my car Toyota RAV4 XLE hybrid 2024. Before buying it, I checked the information regarding compatibility and on the website it was indicated that everything is compatible and works. The installation was done according to the video instructions on the official Comma website. After connecting and the first startup an error popped up that my car is not recognized. I rebooted Comma 3x several times......tried to drive the car but all was unsuccessful. Comma 3x did not work properly + ACC/LKAS errors appeared on the dashboard monitor. After returning all the chips to the place of the factory connection errors disappeared and the car worked in normal mode. photos attached.
Please help me solve my problem. Thank you. Have a nice day.
Perhaps I write with this problem not where it is necessary...... but it is really difficult for me to understand as I do not know English well yet. I am writing this message through a translator. I came to the USA from Ukraine with my family due to the military situation in my country. I apologize in advance for the inconvenience. But please, please help me with this problem. Is it possible to fix it remotely or will I have to return the product?

### Which car does this affect?

Toyota RAV4 XLE hybrid 2024 (VIN: JTME6RFV6RD557044)

### Provide a route where the issue occurs

results: google_104000190100648345810

### openpilot version

0.9.6

### Additional info

I can come over if you need me to? Maybe someone can set me up with the right work. Thank you very much. 
Tried reinstalling the firmware many times nothing helped. Also on the forum https://discord.com/ I was recommended to install some other firmware "installer.comma.ai/commaai/master-ci" but nothing worked. The car is also not recognized. I went back to the original firmware "openpilot.comma.ai". I also drove the car with this firmware in the hope that something will change, but unfortunately nothing changed.

sshane, 2024-04-12:
This does not seem to be a software bug, I see errors reported by panda. I would open a support ticket here: https://comma.ai/support

yahaplus, 2024-04-12:
Hi. Thank you so much for your help and your time.  Yes. There are errors on the dashboard when the unit itself is not connected. What does that mean? 
[image]
[image]
[image]
[image]
[image]
[image]
[image]

yahaplus, 2024-04-12:
> Похоже, это не программная ошибка, я вижу ошибки, о которых сообщила панда, и все встроенное ПО отсутствует. Я бы открыл заявку в службу поддержки здесь: https://comma.ai/support
> 
> Можешь сфотографировать ремень? Есть ли неисправности приборной панели автомобиля, когда кабель USB C comma 3X отсоединен, а жгут проводов подключен?

[image]

yahaplus, 2024-04-12:
> Похоже, это не программная ошибка, я вижу ошибки, о которых сообщила панда, и все встроенное ПО отсутствует. Я бы открыл заявку в службу поддержки здесь: https://comma.ai/support
>
> Можешь [...]

Toyota RAV4 XLE hybrid 2024 (VIN: JTME6RFV6RD557044) purchased from an authorized dealer 3 weeks ago.
The Comma 3X autopilot was purchased two days ago. At the first connection all requirements and instructions were observed......by installed the latest firmware "openpilot.comma.ai". On this firmware error....... that my car is not recognized. 
The following firmware was installed in search of the problem elimination 
https://installer.comma.ai/hardcoded-fp/release3-toyota-rav4-2023
https://installer.comma.ai/commaai/nightly https://installer.comma.ai/commaai/master-ci
All the above firmware did not change anything. My car is still not recognized. 
Also it was found that when the wiring harness is connected according to the requirements in the standard places on the car, there were still errors on ACC/LKAS without connecting the autopilot itself. I don't know if it should be like this or not? Maybe the harness itself was put to me in the wrong place ? Who has the wrong harness? Thank you all very much. Have a good day, everybody.

sshane, 2024-04-13:
I would try a new USB C cable 3.1 gen 2 in the mean time while you are waiting for a response to your support ticket.

yahaplus, 2024-04-13:
What's this cable going to change? Isn't it just a power cable to the autopilot itself? 

Отправлено из Yahoo Почты на iPhone

Пользователь суббота,

nelsonjchen, 2024-04-13:
It's not just power. The cable also handles transporting data as well.

nelsonjchen, 2024-04-13:
Met up with Yan at a local parking lot near where I am living.

First thing we did was swap ODB-C cables. That did not work. My known good cable still caused errors to be thrown.

Then we tried swapping my comma three in place of his comma threex. This also did not work. The errors also showed on my device.

So, we went further up the cables. We swapped the harness box. This made it work! That's a good sign but we continued to swap.

We kept my harness box and installed his harness connector cables. It continued to work. This meant his connector cables are working fine.

And for giggles, we installed his harness box in my Corolla. It causes the same or similar errors to be thrown in my car.

All to say, I believe the harness box is faulty and should be replaced.

arifgarayev, 2024-04-13:
@nelsonjchen are you sure about that replacement?

nelsonjchen, 2024-04-13:
Yes, very.

yahaplus, 2024-04-13:
We swapped everything around and that's the method we used to identify the malfunction. It's all caused by this unit in the photo. Thank you so much Nelsonjchen. 
[image]
[image]
[image]
[image]

yahaplus, 2024-04-13:
[image]
I hope I get this part replaced as soon as possible. Without any problems or complaints to me.

yahaplus, 2024-04-14:
Hi. Yes, 10000000%. 

Отправлено из Yahoo Почты на iPhone

Пользователь суббота,

yahaplus, 2024-04-14:
Yes. Replace that part and everything will work. 

Отправлено из Yahoo Почты на iPhone

Пользователь суббота,

sshane, 2024-04-14:
Make sure to send this thread to our support. Closing here since GitHub is primarily for software issues with openpilot.

closed 2024-04-14 by sshane