commaai/openpilot #32200: Rumbling while collaboratively steering at low speeds

voximity, 2024-04-15:
### Describe the bug

At low speeds (around 15 MPH), my steering wheel will "rumble" if I collaboratively steer alongside lateral input from openpilot. To be precise, this is not the kind of vibration/rumbling that Hondas give the driver to indicate a warning (like lane departure), but almost feels as if I am going over a rumble strip. More accurately, it sounds like I'm fighting the torque actuator -- or something.

### Which car does this affect?

Honda CR-V 2020

### Provide a route where the issue occurs

e4b43f41460a7547/00000001--02564f7b2d/1

### openpilot version

27c26c2 (master-ci)

### Additional info

I initially encountered this issue on Sunnypilot when using its separated LKAS / ACC feature. My CR-V's stock ACC does not allow me to set the cruise below 25 MPH, so I was able to reproduce this on stock OP using OP long and setting my max speed to 15 MPH. In the provided route, the issue happens a lot as I zig-zag through a street near the very end (immediately preceding the bookmarks at the end).

*Possibly* related to #29839, except this feels *and* sounds like a rumble, rather than just being an electric humming sound.

adeebshihadeh, 2024-08-09:
Still an issue?

voximity, 2024-08-09:
> Still an issue?

Yes - not sure if it's something with my particular car, though. I forgot to reach out to another CR-V owner in the Discord and ask about this problem