commaai/openpilot #32242: Toyota: rare inconsistencies in FW response timing

sshane, 2024-04-17:
On very rare occasions, we will time out the eps because it is taking too long to respond (on the order of ~100ms, and our timeout is 100ms). This is commonly after it has already responded to the beginning of the query:

```python
    Request(
      [StdQueries.TESTER_PRESENT_REQUEST, StdQueries.DEFAULT_DIAGNOSTIC_REQUEST, StdQueries.EXTENDED_DIAGNOSTIC_REQUEST, StdQueries.UDS_VERSION_REQUEST],
      [StdQueries.TESTER_PRESENT_RESPONSE, StdQueries.DEFAULT_DIAGNOSTIC_RESPONSE, StdQueries.EXTENDED_DIAGNOSTIC_RESPONSE, StdQueries.UDS_VERSION_RESPONSE],
      ...
      ),
```

so a potential solution could be to double the timeout we add after we get a successful response from any ECU.

We don't want to double the global timeout as that would double the time taken for a lot of the queries that we rely on timing out (VIN query, tester present ECU scan query, not perfectly whitelisted requests, etc.).

adeebshihadeh, 2024-04-17:
Adding to the milestone. We have to fix ASAP.

sshane, 2024-04-17:
Sounds good. The first solution could handle everything and is the easiest and most generic, so I'll get that in first.

---

TODO:
- [ ] Apply temporal checks to first frame responses in IsoTpMessage. This is similar to the consecutive frame ID assertion that disallows just getting infinite consecutive frames from a bad ECU
- [ ] If the above doesn't fix all cases, double the timeout on receive of any valid response. This works because we preface all Toyota queries with tester present+

closed 2024-04-18 by sshane