commaai/openpilot #32347: Build failed on PC (container and native)

ebellumat, 2024-05-03:
### Describe the bug

Hello, this is my first time trying to run Openpilot to contribute, but i'm getting some erros even following the tutorials.

Using the `./devcontainer` environment, I have this issue on the master branch.

Trying to run Openpilot on a container using the command ./launch_openpilot.sh  , what I did possible wrong? qt_resourceFeatureZlib not being found! I'm using the  .devcontainer  envionment available on the project.

[image]

Output

```bash
   ./launch_openpilot.sh 
/workspaces/openpilot/launch_chffrplus.sh: line 7: /data/openpilot/launch_env.sh: No such file or directory
ln: failed to create symbolic link '/data/pythonpath': No such file or directory
/workspaces/openpilot/launch_chffrplus.sh: line 82: /tmp/launch_log: Permission denied
QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-batman'
scons: Reading SConscript files ...
scons: done reading SConscript files.
scons: Building targets ...
cythonize rednose/helpers/ekf_sym_pyx.pyx
Compiling /workspaces/openpilot/rednose/helpers/ekf_sym_pyx.pyx because it changed.
[1/1] Cythonizing /workspaces/openpilot/rednose/helpers/ekf_sym_pyx.pyx
cd /workspaces/openpilot/selfdrive/controls/lib/lateral_mpc_lib && python3 lat_mpc.py
cd /workspaces/openpilot/selfdrive/controls/lib/longitudinal_mpc_lib && python3 long_mpc.py
cythonize selfdrive/modeld/models/commonmodel_pyx.pyx
Retrieved `selfdrive/modeld/models/supercombo_metadata.pkl' from cache
cythonize selfdrive/modeld/runners/runmodel_pyx.pyx
cythonize selfdrive/modeld/runners/snpemodel_pyx.pyx
Warning: Please note that the following versions of CasADi  are officially supported: 3.5.6 or 3.5.5 or 3.5.4 or 3.5.3 or 3.5.2 or 3.5.1 or 3.4.5 or 3.4.0.
 If there is an incompatibility with the CasADi generated code, please consider changing your CasADi version.
Version 3.6.5 currently in use.
Warning: Please note that the following versions of CasADi  are officially supported: 3.5.6 or 3.5.5 or 3.5.4 or 3.5.3 or 3.5.2 or 3.5.1 or 3.4.5 or 3.4.0.
.
.
.
.
.
/usr/bin/ld: selfdrive/ui/assets.o: in function `qResourceFeatureZlib()':
/workspaces/openpilot/selfdrive/assets/assets.cc:34225: undefined reference to `qt_resourceFeatureZlib'
/usr/bin/ld: /workspaces/openpilot/selfdrive/assets/assets.cc:34225: undefined reference to `qt_resourceFeatureZlib'
clang: error: linker command failed with exit code 1 (use -v to see invocation)
/usr/bin/ld: selfdrive/ui/assets.o: in function `qResourceFeatureZlib()':
/workspaces/openpilot/selfdrive/assets/assets.cc:34225: undefined reference to `qt_resourceFeatureZlib'
/usr/bin/ld: /workspaces/openpilot/selfdrive/assets/assets.cc:34225: undefined reference to `qt_resourceFeatureZlib'
clang: error: linker command failed with exit code 1 (use -v to see invocation)
/usr/bin/ld: selfdrive/ui/assets.o: in function `qResourceFeatureZlib()':
/workspaces/openpilot/selfdrive/assets/assets.cc:34225: undefined reference to `qt_resourceFeatureZlib'
/usr/bin/ld: /workspaces/openpilot/selfdrive/assets/assets.cc:34225: undefined reference to `qt_resourceFeatureZlib'
clang: error: linker command failed with exit code 1 (use -v to see invocation)
/usr/bin/ld: selfdrive/ui/assets.o: in function `qResourceFeatureZlib()':
/workspaces/openpilot/selfdrive/assets/assets.cc:34225: undefined reference to `qt_resourceFeatureZlib'
/usr/bin/ld: /workspaces/openpilot/selfdrive/assets/assets.cc:34225: undefined reference to `qt_resourceFeatureZlib'
/usr/bin/ld: selfdrive/ui/assets.o: in function `qResourceFeatureZlib()':
/workspaces/openpilot/selfdrive/assets/assets.cc:34225: undefined reference to `qt_resourceFeatureZlib'
/usr/bin/ld: /workspaces/openpilot/selfdrive/assets/assets.cc:34225: undefined reference to `qt_resourceFeatureZlib'
scons: *** [selfdrive/ui/qt/setup/updater] Error 1
scons: *** [selfdrive/ui/qt/setup/setup] Error 1
scons: *** [selfdrive/ui/tests/test_translations] Error 1
clang: error: linker command failed with exit code 1 (use -v to see invocation)
scons: *** [selfdrive/ui/ui] Error 1
clang: error: linker command failed with exit code 1 (use -v to see invocation)
scons: *** [selfdrive/ui/tests/ui_snapshot] Error 1
scons: building terminated because of errors.
scons: Reading SConscript files ...
scons: done reading SConscript files.
scons: Building targets ...
clang++ -o rednose/helpers/ekf_sym_pyx.so -pthread -shared -Wl,-rpath=/workspaces/openpilot/third_party/acados/x86_64/lib -Wl,-rpath=/workspaces/openpilot/third_party/snpe/x86_64 rednose/helpers/ekf_sym_pyx.o rednose/helpers/ekf_load.os rednose/helpers/ekf_sym.os -Lthird_party/acados/x86_64/lib -Lthird_party/libyuv/x86_64/lib -L/usr/lib -L/usr/local/lib -Lthird_party/snpe/x86_64 -Lcereal -Lthird_party -Lopendbc/can -Lselfdrive/boardd -Lcommon -Lrednose/helpers -ldl common/libcommon.a -ljson11 -lzmq
clang -o selfdrive/controls/lib/lateral_mpc_lib/c_generated_code/acados_ocp_solver_pyx.o -c -std=gnu11 -DSWAGLOG="\"common/swaglog.h\"" -g -fPIC -O2 -Wunused -Wshadow -Wno-unknown-warning-option -Wno-deprecated-register -Wno-register -Wno-inconsistent-missing-override -Wno-c99-designator -Wno-reorder-init-list -Wno-error=unused-but-set-variable -Wno-vla-cxx-extension -DSWAGLOG="\"common/swaglog.h\"" -Wno-#warnings -Wno-shadow -Wno-deprecated-declarations -I. -Ithird_party/acados/include -Ithird_party/acados/include/blasfeo/include -Ithird_party/acados/include/hpipm/include -Ithird_party/catch2/include -Ithird_party/libyuv/include -Ithird_party/json11 -Ithird_party/linux/include -Ithird_party/snpe/include -Ithird_party/qrcode -Ithird_party -Icereal -Iopendbc/can -Ithird_party/maplibre-native-qt/include -Ithird_party/maplibre-native-qt/x86_64/include -I/home/batman/pyenv/versions/3.11.4/include/python3.11 -I.venv/lib/python3.11/site-packages/numpy/core/include selfdrive/controls/lib/lateral_mpc_lib/c_generated_code/acados_ocp_solver_pyx.c
[... 41 lines cut ...]
/usr/bin/ld: third_party/maplibre-native-qt/x86_64/lib/libQMapLibre.so: undefined reference to `ubidi_close_66'
/usr/bin/ld: third_party/maplibre-native-qt/x86_64/lib/libQMapLibre.so: undefined reference to `ubidi_countParagraphs_66'
scons: *** [selfdrive/ui/_spinner] Error 1
/usr/bin/ld: third_party/maplibre-native-qt/x86_64/lib/libQMapLibre.so: undefined reference to `u_errorName_66'
clang: error: linker command failed with exit code 1 (use -v to see invocation)
scons: *** [selfdrive/ui/_text] Error 1
/usr/bin/ld: warning: libicuuc.so.66, needed by third_party/maplibre-native-qt/x86_64/lib/libQMapLibre.so, not found (try using -rpath or -rpath-link)
/usr/bin/ld: third_party/maplibre-native-qt/x86_64/lib/libQMapLibre.so: undefined reference to `ubidi_writeReverse_66'
/usr/bin/ld: third_party/maplibre-native-qt/x86_64/lib/libQMapLibre.so: undefined reference to `u_shapeArabic_66'
/usr/bin/ld: third_party/maplibre-native-qt/x86_64/lib/libQMapLibre.so: undefined reference to `ubidi_getParagraphByIndex_66'
/usr/bin/ld: third_party/maplibre-native-qt/x86_64/lib/libQMapLibre.so: undefined reference to `ubidi_setLine_66'
/usr/bin/ld: third_party/maplibre-native-qt/x86_64/lib/libQMapLibre.so: undefined reference to `ubidi_writeReordered_66'
/usr/bin/ld: third_party/maplibre-native-qt/x86_64/lib/libQMapLibre.so: undefined reference to `ubidi_open_66'
/usr/bin/ld: third_party/maplibre-native-qt/x86_64/lib/libQMapLibre.so: undefined reference to `ubidi_countRuns_66'
/usr/bin/ld: third_party/maplibre-native-qt/x86_64/lib/libQMapLibre.so: undefined reference to `ubidi_getProcessedLength_66'
/usr/bin/ld: third_party/maplibre-native-qt/x86_64/lib/libQMapLibre.so: undefined reference to `ubidi_setPara_66'
/usr/bin/ld: third_party/maplibre-native-qt/x86_64/lib/libQMapLibre.so: undefined reference to `ubidi_getVisualRun_66'
/usr/bin/ld: third_party/maplibre-native-qt/x86_64/lib/libQMapLibre.so: undefined reference to `ubidi_close_66'
/usr/bin/ld: third_party/maplibre-native-qt/x86_64/lib/libQMapLibre.so: undefined reference to `ubidi_countParagraphs_66'
/usr/bin/ld: third_party/maplibre-native-qt/x86_64/lib/libQMapLibre.so: undefined reference to `u_errorName_66'
clang: error: linker command failed with exit code 1 (use -v to see invocation)
scons: *** [selfdrive/ui/installer/installers/installer_openpilot] Error 1
scons: building terminated because of errors.

`

ebellumat, 2024-05-04:
I Just removed the Pop OS from my laptop and installed Ubuntu 20.04 LTS, working just fine.

closed 2024-05-04 by adeebshihadeh

JohnNiang, 2024-05-20:
I encountered the same issue in Ubuntu 22.04.4.

```bash
parallels@ubuntu:~$ cat /etc/os-release 
PRETTY_NAME="Ubuntu 22.04.4 LTS"
NAME="Ubuntu"
VERSION_ID="22.04"
VERSION="22.04.4 LTS (Jammy Jellyfish)"
VERSION_CODENAME=jammy
ID=ubuntu
ID_LIKE=debian
HOME_URL="https://www.ubuntu.com/"
SUPPORT_URL="https://help.ubuntu.com/"
BUG_REPORT_URL="https://bugs.launchpad.net/ubuntu/"
PRIVACY_POLICY_URL="https://www.ubuntu.com/legal/terms-and-policies/privacy-policy"
UBUNTU_CODENAME=jammy
```

```bash
parallels@ubuntu:~$ sudo apt-get install libicu-dev
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
libicu-dev is already the newest version (70.1-2).
0 upgraded, 0 newly installed, 0 to remove and 3 not upgraded.
```

muzi2045, 2024-05-23:
Ubuntu 24.04 LTS same issue here

> /usr/bin/ld: third_party/maplibre-native-qt/x86_64/lib/libQMapLibre.so: undefined reference to `ubidi_writeReverse_66'
> /usr/bin/ld: third_party/maplibre-native-qt/x86_64/lib/libQMapLibre.so: [...]

PinkGabriel, 2024-05-24:
fixed this by:
```
wget http://security.ubuntu.com/ubuntu/pool/main/i/icu/libicu66_66.1-2ubuntu2_amd64.deb
sudo dpkg -i libicu66_66.1-2ubuntu2_amd64.deb

```
to provide libicuuc.so.66