commaai/openpilot #32350: Simulator Bridge PLATFORM_NOT_FOUND_KHR

devimauz, 2024-05-04:
### Describe the bug

Hi,
Sadly can't run SIM in VM.
I'm not sure its graphics problem because UI runs.
Thanks.

```
[INFO] Environment: MetaDriveEnv
[INFO] MetaDrive version: 0.4.2.3
[INFO] Sensors: [lidar: Lidar(), side_detector: SideDetector(), lane_line_detector: LaneLineDetector(), rgb_road: RGBCameraRoad(1928, 1208)]
[INFO] Render Mode: offscreen
[INFO] Horizon (Max steps per agent): None
[INFO] Assets version: 0.4.2.3
[INFO] Known Pipes: glxGraphicsPipe
[WARNING] You are using too large buffer! The height is 1208, and width is 1928. It may lower the sample efficiency! Consider reducing buffer size or use cuda image by set [image_on_cuda=True]. (base_camera.py:46)
[INFO] Start Scenario Index: 0, Num Scenarios : 1
[WARNING] env.vehicle will be deprecated soon. Use env.agent instead (base_env.py:719)
Starting listener for: camerad
Process bridge:
Traceback (most recent call last):
  File "/home/akkount/.pyenv/versions/3.11.4/lib/python3.11/multiprocessing/process.py", line 314, in _bootstrap
    self.run()
  File "/home/akkount/.pyenv/versions/3.11.4/lib/python3.11/multiprocessing/process.py", line 108, in run
    self._target(*self._args, **self._kwargs)
  File "/home/akkount/openpilot/openpilot/tools/sim/bridge/common.py", line 58, in bridge_keep_alive
    self._run(q)
  File "/home/akkount/openpilot/openpilot/tools/sim/bridge/common.py", line 89, in _run
    self.simulated_sensors = SimulatedSensors(self.dual_camera)
                             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/akkount/openpilot/openpilot/tools/sim/lib/simulated_sensors.py", line 19, in __init__
    self.camerad = Camerad(dual_camera=dual_camera)
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/akkount/openpilot/openpilot/tools/sim/lib/camerad.py", line 28, in __init__
    self.ctx = cl.create_some_context()
               ^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/akkount/openpilot/.venv/lib/python3.11/site-packages/pyopencl/__init__.py", line 1602, in create_some_context
    platforms = get_platforms()
                ^^^^^^^^^^^^^^^
pyopencl._cl.LogicError: clGetPlatformIDs failed: PLATFORM_NOT_FOUND_KHR
Stopping listener for: camerad
Traceback (most recent call last):
  File "/home/akkount/openpilot/tools/sim/./run_bridge.py", line 43, in <module>
    keyboard_poll_thread(queue)
  File "/home/akkount/openpilot/openpilot/tools/sim/lib/keyboard_ctrl.py", line 59, in keyboard_poll_thread
    c = getch()
        ^^^^^^^
  File "/home/akkount/openpilot/openpilot/tools/sim/lib/keyboard_ctrl.py", line 47, in getch
    ch = sys.stdin.read(1)
         ^^^^^^^^^^^^^^^^^
KeyboardInterrupt

```

### OS Version

20.04

StefanLekanic-SYRMIA, 2024-05-10:
Try this: https://stackoverflow.com/questions/11990131/python-logicerror-clgetplatformids-failed-platform-not-found-khr

adeebshihadeh, 2024-05-19:
Our issue tracker is not for help getting started; check out the community Discord instead: https://discord.comma.ai.

closed 2024-05-19 by adeebshihadeh