commaai/openpilot #32351: Can error check connection - iqonic 5 2023

wanderer, 2024-05-04:
### Describe the bug

I installed open pilot and I get the message "Can error check connection". Nothing seems to work. I have turning the USB cable around. 

### Which car does this affect?

Hyundai iqonic 5 2023

### Provide a route where the issue occurs

2fa352365f86a9b2|2024-05-04--16-17-56

### openpilot version

0.9.6

sshane, 2024-05-05:
Can you try this branch? The bus structure looks like an HDA 2 variant, do you know if they market HDA and HDA 2 in your region? And which do you have?

https://installer.comma.ai/commaai/ioniq-5-fix

wanderer, 2024-05-06:
@sshane thank you for the quick replay. I have a UK 2023 Ioniq. After digging around I read that I probably don't have HDA2. (from this post https://www.reddit.com/r/Ioniq5/comments/1bcwf16/hda_15_on_a_top_spec_2023/) I might have HDA "1.5". But I have not figured out how to tell exactly.

wanderer, 2024-05-06:
As for ADAS Features i think I have
- Driving Convenience
- Forward Safety
- DAW
- Parking Safety
- Blind-spot Safety

[image]
[image]
[image]
[image]

sshane, 2024-05-06:
I believe you ordered the Hyundai Q harness which is meant for HDA2 vehicles, but your bus structure looks to be more similar to HDA. You will want to contact support or open a return to get the Hyundai K harness that is meant for non-HDA2 variants.

Once you have that harness, the missing camera message isn't a requirement as that's only for HDA2.

As an alternate option (and at your own risk), some community members have swapped the wires for CAN 1 and CAN 2 so you won't need to return the harness. See the diagrams for both:

https://github.com/commaai/neo/blob/master/car_harness/Sonata_Harness.sch.pdf

https://github.com/commaai/neo/blob/master/car_harness/EV6_Harness.sch.pdf

closed 2024-05-06 by sshane