commaai/openpilot #32353: 2024 Elantra car Unrecognized

1okko, 2024-05-06:
### Describe the bug

2024 Elantra car Unrecognized

### Which car does this affect?

2024 HYUNDAI Elantra 

### Provide a route where the issue occurs

d01e5725e5587ae3/00000001--a4126f0d93/0

### openpilot version

0.9.7

### Additional info

It can be confirmed that the vehicle is a CAN and a camera SCC

sshane, 2024-05-06:
Has a CAN FD radar bus. Please try this branch: https://github.com/commaai/openpilot/tree/cn7c-elantra

1okko, 2024-05-07:
ok.i will try and let you know.

1okko, 2024-05-08:
d01e5725e5587ae3/00000000--178533cb2e/1

it works well，thanks

1okko, 2024-05-08:
> 拥有CAN FD雷达总线。请尝试这个分支：https://github.com/commaai/openpilot/tree/cn7c-elantra

d01e5725e5587ae3/00000000--178533cb2e/1

效果很好，谢谢

1okko, 2024-05-08:
> d01e5725e5587ae3/00000000--178533cb2e/1
> 
> it works well，thanks

d01e5725e5587ae3/00000000--178533cb2e/1

it works well，thanks

sshane, 2024-05-08:
Nice! Can you drive more to gather more lateral control data?

1okko, 2024-05-08:
> d01e5725e5587ae3/00000000--178533cb2e/1

d01e5725e5587ae3/00000000--178533cb2e/1  this not enough？

sshane, 2024-05-09:
Need some more curves I believe!

1okko, 2024-05-09:
> Need some more curves I believe!

ok， i will drive it today

1okko, 2024-05-10:
> Need some more curves I believe!

d01e5725e5587ae3/00000002--3ec9e0ec2a/0

sshane, 2024-05-11:
High speed (> 80 km/h) and engaged curves I mean

sshane, 2024-05-22:
@1okko can you get more driving data please? Then we can merge!

1okko, 2024-05-25:
> @1okko can you get more driving data please? Then we can merge!

d01e5725e5587ae3/00000003--ecb5e1500a/2

sshane, 2024-05-25:
| Dongle, platform, VIN                                                                                                           | Name from VIN 🆚 CarDocs   | Segments                                | Bad seg tags   | Good seg tags                                         |
|:--------------------------------------------------------------------------------------------------------------------------------|:---------------------------|:----------------------------------------|:---------------|:------------------------------------------------------|
| [d01e5725e5587ae3](https://useradmin.comma.ai/?onebox=d01e5725e5587ae3)<br><sub>HYUNDAI_ELANTRA_2021<br>000000000........</sub> | None 🆚<br>None            | 4 routes,<br>94 segments<br>(1.6 hours) |                | - all lat active: 51<br>- no input all lat active: 42 |

sshane, 2024-05-25:
Need a little more sharper turns at high speeds, the torqued learner doesn't have many data points there.

1okko, 2024-05-25:
> Need a little more sharper turns at high speeds, the torqued learner doesn't have many data points there.

ok

1okko, 2024-06-01:
> Need a little more sharper turns at high speeds, the torqued learner doesn't have many data points there.

d01e5725e5587ae3/00000004--9eae7430ad/0

sshane, 2024-06-04:
| Dongle, platform, VIN                                                                                                           | Name from VIN 🆚 CarDocs   | Segments                                 | Bad seg tags                                                                                                 | Good seg tags                                                                      |
|:--------------------------------------------------------------------------------------------------------------------------------|:---------------------------|:-----------------------------------------|:-------------------------------------------------------------------------------------------------------------|:-----------------------------------------------------------------------------------|
| [d01e5725e5587ae3](https://useradmin.comma.ai/?onebox=d01e5725e5587ae3)<br><sub>HYUNDAI_ELANTRA_2021<br>000000000........</sub> | None 🆚<br>None            | 6 routes,<br>181 segments<br>(3.0 hours) | - [high lateral accel factor: 1](https://useradmin.comma.ai/?onebox=d01e5725e5587ae3%7C00000005--6ecc905ec8) | - all lat active: 87<br>- no input all lat active: 57<br>- valid torque params: 78 |

sshane, 2024-06-04:
I forgot this port requires changing the length of `LFAHDA_MFC`, we'll have to put this off for now until we have for a car porting sprint. Feel free to keep driving on the branch!

1okko, 2024-06-15:
> I forgot this port requires changing the length of `LFAHDA_MFC`, we'll have to put this off for now until we have time for a car porting sprint. Feel free to keep driving on the branch!

How can I test this fork on c3+red panda kit?