commaai/openpilot #32386: Ship Ubuntu 24.04 + mainline kernel to master

adeebshihadeh, 2024-05-09:
Now that the initial 24.04 work is done, two hard parts remain: mainlining the kernel and doing the validation to get both shipped to master.

`test_onroad` is a good litmus test for functionality while developing. Once it's in a state where the test passes reliably on a single device, I can deploy it to to several devices in our internal testing closet. 

Locked to @andiradulescu, who did the initial 24.04 work.

muzi2045, 2024-05-24:
Ubuntu 24.04 LTS + CUDA 12.5 failed in modeld. 
the pre-build onnxruntime lib link to wrong cuda libs. (so wired)
[image]

adeebshihadeh, 2024-08-11:
FYI, @deanlee the mainline kernel has a different driver and API for the camera. All camerad work should focus on making that port easy.

deanlee, 2024-08-12:
> FYI, @deanlee the mainline kernel has a different driver and API for the camera. All camerad work should focus on making that port easy.

got it

adeebshihadeh, 2024-08-14:
@andiradulescu I consider 24.04 since it's merged into agnos-builder master. No reason to ship an update to openpilot master yet. I think we'll do that once the image is smaller.

workinright, 2025-07-16:
Hi

Is the kernel mainlining part also still locked, or can I start working on this?

adeebshihadeh, 2025-07-16:
> Hi
> 
> Is the kernel mainlining part of the bounty also still locked, or can I start working on this?

Unlocked -- good luck!

workinright, 2025-07-16:
Thank you @adeebshihadeh .

The only thing is that despite having a stack of SDM845 devices in my lab, I don't have any Comma device. Could anybody possibly share me one remotely? An SSH tunnel to the device would be the bare minimum, USB/IP or other connection enabling me to use the EDL mode would be nice, UART exposed would be perfect.

Also, what is the reason you're not using the SDM845's integrated modem? Because of the regulatory reasons? Lacking RF/PA components on the board? Both?

adeebshihadeh, 2025-07-16:
> Also, what is the reason you're not using the SDM845's integrated modem? Because of the regulatory reasons? Lacking RF/PA components on the board? Both?

Two reasons: the SOMs we originally used were SDAs but the primary reason now is cost (license, regulatory, etc).

workinright, 2025-07-16:
Okay, sure enough, using an external ready-to-use module bundling everything in one is very tempting indeed :) Is the main SoC's internal GPS used (in which case you still need to run the baseband firmware on the modem Hexagon DSP core), or you also use the external Quectel module's GPS?