commaai/openpilot #32422: Take Control Immediately, CAN BUS Disconnected and Take Control, Controls are Lagging

crwayman7, 2024-05-13:
### Describe the bug

  I faced these issues before in other forks. I was discussing these issues with Sunny here: https://discord.com/channels/469524606043160576/538741329799413760/1237469523297173555

  He recommended to Install Stock OpenPilot and open a git issue if I was able to replicate on stock OpenPilot, which I was able to do so. 

   I am not sure if this vechicle specific or not, however, I drive a 2023 Kia Sorento PHEV.  There are two errors I see occasionally. One seems to be just a quick message and OpenPilot seems to be functional. The message appears, then goes away after about 5-10 seconds which has an orange/yellow background and states to "Take Control, Controls are Lagging." 

  The other message seems more crucial and it feels like OpenPilot loses control of the vehicle. The message has a red background and states "Take Control Immediately, CAN BUS Disconnected." 

### Provide a route where the issue occurs

https://connect.comma.ai/8f755f62fd10cbc0/1715441744631/1715445504266

### openpilot version

0.9.6-release / release3 / e5d2ce5 / Feb 27

### Additional info

  If my dongle ID is needed, just ask.  I read the rules about contributing here. I will spin up an Ubuntu VM tonight and create the dev environment. I work data storage so I only have RHEL and Fedora but have used Ubuntu before, so should be no issue.

sshane, 2024-05-14:
I requested upload of that segment, can you put the device on WiFI?

adeebshihadeh, 2024-05-14:
Thanks for reporting. Almost certainly the same as #32286, so we'll track it there. We're actively working on fixing this.

closed 2024-05-14 by adeebshihadeh

crwayman7, 2024-05-14:
> I requested upload of that segment, can you put the device on WiFi?

@sshane apologies, I did take it offline temporarily. It's on WiFi now. It's the same dongle ID I DM'd you a few weeks ago. I will DM again.

crwayman7, 2024-05-26:
I just did a 10 hour drive yesterday 25 May and saw not only these CAN errors i describe in  this issue but other new ones that I've seen as well. A lot of unexpected and random disengagements. All on stock OpenPilot if y'all want to check it out.

adeebshihadeh, 2024-05-26:
> I just did a 10 hour drive yesterday 25 May and saw not only these CAN errors i describe in this issue but other new ones that I've seen as well. A lot of unexpected and random disengagements. All on stock OpenPilot if y'all want to check it out.

Can you post the route?

crwayman7, 2024-05-27:
Here are the routes

https://connect.comma.ai/8f755f62fd10cbc0/1716633995841/1716646549925

https://connect.comma.ai/8f755f62fd10cbc0/1716646826698/1716656304031

https://connect.comma.ai/8f755f62fd10cbc0/1716656977790/1716669859155

crwayman7, 2024-05-27:
> > I just did a 10 hour drive yesterday 25 May and saw not only these CAN errors i describe in this issue but other new ones that I've seen as well. A lot of unexpected and random disengagements. All on stock OpenPilot if y'all want to check it out.
> 
> Can you post the route?

Here are the routes

https://connect.comma.ai/8f755f62fd10cbc0/1716633995841/1716646549925

https://connect.comma.ai/8f755f62fd10cbc0/1716646826698/1716656304031

https://connect.comma.ai/8f755f62fd10cbc0/1716656977790/1716669859155

waymo

adeebshihadeh, 2024-05-30:
Thanks for sending those over. I'm still working on this, but I put up a branch (`boardd-spi-test`) if you wanna try the fixes early (use the URL `installer.comma.ai/commaai/boardd-spi-test`). I'm writing a stress test that will run in CI and validate all the fixes.

crwayman7, 2024-05-30:
> @crwayman7 Thanks for sending those over. I'm still working on this, but I put up a branch (`boardd-spi-test`) if you wanna try the fixes early (use the URL `installer.comma.ai/commaai/boardd-spi-test`). I'm writing a stress test that will run in CI and validate all the fixes.

Good thing you hit me up today. I am actually driving home tomorrow, so another 10 hours. I will install the branch now and update this issue tomorrow after the drive with the route information.

Waymo

crwayman7, 2024-06-01:
> @crwayman7 Thanks for sending those over. I'm still working on this, but I put up a branch (`boardd-spi-test`) if you wanna try the fixes early (use the URL `installer.comma.ai/commaai/boardd-spi-test`). I'm writing a stress test that will run in CI and validate all the fixes.

  I had fewer issues with this branch than I did with stock OpenPilot. Still, some issues were present "CAN Bus Disconnected" "Controls Laggy" and something like "Communication Between Processes," but they happened less frequently, and again, this was more stable. Previously you could expect it every 10 minutes or so. With this one, I had stability for 30 minutes to an hour in some instances. Here are the routes:

https://connect.comma.ai/8f755f62fd10cbc0/1717154570506/1717166410435

https://connect.comma.ai/8f755f62fd10cbc0/1717168077800/1717179380390

https://connect.comma.ai/8f755f62fd10cbc0/1717179844128/1717193562336

waymo