commaai/openpilot #32463: Replace PlotJuggler with Rerun

adeebshihadeh, 2024-05-18:
We've got a proof of concept of https://rerun.io support in `tools/rerun/`. Rerun will support all the things we want out of a visualization tool. The first step is to replace PlotJuggler, which we mostly use for ad-hoc debugging. Once this is done, we can do cool things like showing all three cameras alongside your plots.

At a minimum, this means:
* demo route fully loads in <10s
* rerun runs smoothly (>30 fps). the POC seems to have a leak and it's 10 fps on my machine
* can plot all the same things as PlotJuggler, with the same workflow
* nice README, similar to PlotJuggler's

Community feedback/discussion is at https://github.com/commaai/openpilot/discussions/32419.

snigdha510, 2024-05-22:
please assign this to me

d80ep08th, 2024-06-06:
Is this closed?

mahdiait, 2024-07-05:
It looks like you're referring to a bounty or task for replacing PlotJuggler with Rerun in a specific context or issue (possibly issue #32463). To get this done, you might want to follow these steps:

1. **Understand Requirements**: Make sure you know exactly what PlotJuggler is currently used for and what Rerun should replace or improve. Review the documentation or specifications for both tools.

2. **Assess Dependencies**: Check if there are any dependencies or integrations tied to PlotJuggler that need to be updated for Rerun.

3. **Implementation**: Begin the process of replacing PlotJuggler with Rerun in the codebase. This might involve updating scripts, modifying configurations, or adapting existing workflows.

4. **Testing**: Ensure thorough testing is performed to verify that Rerun works as expected in place of PlotJuggler.

5. **Documentation**: Update any related documentation to reflect the change from PlotJuggler to Rerun.

6. **Submit Changes**: Prepare and submit a pull request or change request, addressing issue #32463, and make sure to follow any contribution guidelines for the project.

If you need detailed instructions or assistance with a specific part of this process, feel free to ask!

adeebshihadeh, 2024-07-24:
Locked to @bongbui321, pending Rerun https://github.com/rerun-io/rerun/milestone/30 due to perf issues in their current version

sshane, 2024-08-16:
Some things I use quite often that we want in rerun:

- [ ] custom live functions (ability to reference multiple signals, time shift, etc)
- [ ] layouts: save, load
- [ ] show raw points easily
- [ ] drag and drop
- [ ] signal searching
- [ ] easily manipulate plots: can add new signals on top or a new plot, split horizontally/vertically, zooming effects all plots at once, can see raw values via scrubbar
- [ ] better performance, rerun's UI runs at a lower FPS for me
- [ ] demo route never fully loads, or unloads and takes a while to load

ntegan1, 2024-12-05:
[image]

in 0.18.0 new ingestion method preferred to rr.log for timeseries: rr.send_columns

ntegan1, 2025-01-19:
It seems the main thing holding rerun back from being a plotjuggler contender is custom live functions / transforms (deriv/integral/runningaverage)  as well as raw h265 video support.

the re-mp4 rust crate could be improved to support our h265 hevc videos, but we could easily do the same in plotjuggler.  
plotjuggler videoviewer plugin depends on libqtav-dev which is nonexistant in ubuntu 24.10, wang-bin discontinued it in favor of their mdk-sdk repo.

ntegan1, 2025-01-23:
current plotjuggler VideoViewer plugin depends on QtAV which is deprecated in favor of wang-bin's mdk-sdk.  
for our videos, all it does is decode our video into frames and then compress via qoi.h and store in memory via array/vector.  
This allows for fine-grained frame presentation based on our log data, one of our time series converts PJ `absolute_time` aka logmonotime to frame number which tells the VideoViewer plugin when to present each decoded frame.

Ubuntu 24.10 doesn't have libqtav-dev, and it doesn't look like fun to try to get it to compile with ffmpeg 6/7. QtAV uses a lot of deprecated autoresampling / libswsresample APIs.

I made a proof of concept replacing QtAV usage with qtmultimedia and qtmultimediawidget mediaplayer (see commaai/PJ pulls). The downside of this is that it just pauses and starts playback of our h265 video like any other media player (after calling media player setPosition to the PJ statepublished timestamp) rather than seeking to each individual frame like the QtAV solution did by decoding and manually presenting frames.

dasmith656, 2025-05-06:
👋

adeebshihadeh, 2025-08-09:
Unfortunately, Rerun is still too slow, clunky, and doesn't have the feature set we need. Moving on to https://github.com/commaai/openpilot/issues/35942.

closed 2025-08-09 by adeebshihadeh