commaai/openpilot #32504: Remove pyenv

adeebshihadeh, 2024-05-21:
Rationale
* openpilot itself shouldn't care too much about which Python it runs
* since pyenv builds python from source, it adds a bunch of dependencies that get lumped in with openpilot's own dependencies
* users should choose how to manage their python env

I think this will be doable once we're on 24.04, which uses Python 3.12. AGNOS may jump to future Python versions for speedups, but openpilot should be mostly agnostic to version for core functionality.

adeebshihadeh, 2024-05-22:
An idea for doing this now on 20.04:
* for CI/docker, install 3.11 using the deadsnakes ppa, then update `python3` in the docker to point to 3.11
* users can choose how they want 3.11 installed. they can run our install script used in CI for deadsnakes or use something else like pyenv

Then, going forward, we don't care about Python version aside from >= 3.11.

gregjhogan, 2024-05-23:
This is a super fast way to get a specific version of python installed in GitHub actions if that is useful. I believe they prebuild and slim down every version for a few architectures. 
https://github.com/actions/setup-python

BBBmau, 2024-05-23:
> This is a super fast way to get a specific version of python installed in GitHub actions if that is useful. I believe they prebuild and slim down every version for a few architectures. https://github.com/actions/setup-python

should we do this now or wait until we're on 24.04?

closed 2024-05-23 by adeebshihadeh