commaai/openpilot #32516: [$250 bounty] Update to Python 3.12

adeebshihadeh, 2024-05-22:
This is a pre-req to updating to Ubuntu 24.04.

Here's my last attempt: https://github.com/commaai/openpilot/pull/32296.

Main blocker is upstreaming 3.12 support to MetaDrive: https://github.com/metadriverse/metadrive.

BBBmau, 2024-05-23:
@adeebshihadeh requesting this to be locked since I have a PR on upstream metadrive to support 3.12:
https://github.com/metadriverse/metadrive/pull/720

I'll work on the 3.12 update for openpilot soon.

BBBmau, 2024-05-26:
isn't another major blocker coming from this not being merged? https://github.com/vpelletier/pprofile/pull/51

Doesn't seem like it'll be happening anytime soon.

adeebshihadeh, 2024-05-26:
We can just remove it for now. We don’t use it that much.

PR #32548 (2024-06-11): Update to Python 3.12

Fixes #32516

Second PR was created due to the revert of `Removing Pyenv`

Same as the first PR, it uses a custom pypackage that uses metadrive with py3.12 support in order to get this merged now. Once the official metadrive repo updates and publishes a new release that supports 3.12 we can update it to use that one instead of this custom one.

https://pypi.org/project/metadrive-simulator-py3-12/
closed 2024-06-11 by maxime-desroches