commaai/openpilot #32519: master-ci: Display brightness not adapting to scene properly/Camera gain seems high and dark areas have high noise on display.

nabeelr, 2024-05-23:
### Describe the bug

Display is very bright at night, despite the scene being very dark. In comparison to release3, master-ci from "1716286461 2024-05-21 10:14:21 +0000" seems to not adapt display brightness with regard to the environment very well, where-as older versions of master-ci and release3 seem to handle it very well.

It also seems that there is more noise in the dark spots of the video feed shown on the display with master-ci, though this may be a factor of the brighter display making the noise that was always there more noticeable.

My understanding is that camera gain is used to determine screen brightness. Is it possible the camera gain was tweaked but how screen brightness is derived from camera gain was not?

### Provide a route where the issue occurs

dd154b34dc4d8c52/000000b4--075f0bb175

### openpilot version

master-ci | e42f03e95870cc317b318388beb2efb2b7dfba10 | 1716286461 2024-05-21 10:14:21 +0000

adeebshihadeh, 2024-11-13:
Should be fixed, #33041

closed 2024-11-13 by adeebshihadeh

nabeelr, 2024-11-13:
Thanks Adeeb!