commaai/openpilot #32522: [$100 bounty] Switch to uv package manager

adeebshihadeh, 2024-05-23:
Switch the default package manager from poetry to uv. uv is quite new, so it might be not be ready for us yet. Let's try it out though.

https://github.com/astral-sh/uv

hernantech, 2024-05-24:
Requesting lock on this bounty with a [draft PR](https://github.com/commaai/openpilot/pull/32527). Exporting req's and setting them up in uv then testing validity and performance. Need to migrate checks to uv as well.

adeebshihadeh, 2024-05-28:
Unfortunately, we can't switch quite yet due to the lack of locking, but @signed-long's PR is a sizable speedup. Looking forward to switching once it has multi platform locking!

closed 2024-05-28 by adeebshihadeh

signed-long, 2024-05-28:
Subbed to https://github.com/astral-sh/uv/issues/3350, if I hear anything I'll shout