commaai/openpilot #32525: Hyundai: detect lateral API

sshane, 2024-05-24:
A lot of newer Hyundai vehicles we have yet to merge are blocked on different lateral control behavior, a mostly complete list is here: https://github.com/commaai/openpilot/discussions/30810

We should find a way to detect when the API is different. For the cars above, the main two differences we care about are achieving significantly more or less lateral acceleration when applying the same torque.

To prevent fingerprinting on these new vehicles and giving our users a bad experience, our fuzzy fingerprinting function checks that we've seen the camera part number and build/flash date in the FW. This, however, stops a lot of new cars from fingerprinting when the API has not changed, as we don't know if anything has changed when one of these values gets bumped.

For example, the 2024 EV6 seems the same as previous models with a new part number: https://github.com/commaai/openpilot/pull/32524 However the 2024 Kia Sorento Hybrid acheives significantly less turning acceleration (less than half) and the only thing that bumped was the date in the FW by a few months (which I suspect is either part build date, FW build date, or flash date).

estiens, 2024-05-25:
2024 Ioniq 5 (HDA2) seems to have different wiring harness, not sure if different behavior

sshane, 2024-05-25:
Ioniq 5 N or Ioniq 5?