commaai/openpilot #32531: GM Camera ACC ICE: ECM respects different brake signal than camera

sshane, 2024-05-25:
This is a ***stock ACC*** route where the user pressed the brake and caused an ACC fault we alerted for. It appears that the camera uses what we found for the Bolt, a single bit signal: `ECMEngineStatus->BrakePressed` as it stayed engaged until the ECM faulted. However the ECM might look at `ECMAcceleratorPos->BrakePedalPos` or `EBCMBrakePedalPosition->BrakePedalPosition`

I performed extensive tests to find an appropriate brake signal for the Bolt, so this looks like it might only apply for ICE GMs.

[image]

dc7716b32bf25574/2024-05-24--18-30-20

garrettpall, 2024-08-19:
I’m having the same issue with my SDGM car with OP long enabled (using custom hardware). I will do some testing this week

garrettpall, 2024-08-19:
Using `ret.brakePressed = ret.brake >= 8` seemed to fix the issue with my car. Will continue to test this week