commaai/openpilot #32536: [$250 bounty] Speedup `test_car_interfaces`

adeebshihadeh, 2024-05-25:
Takes ~1-2s per car. That's a big hit since we support nearly 300 different cars.

Goal is <=0.2s avg and <1s max on the CI machine, with the same test coverage.

Command to run:
```
pytest --durations=0 selfdrive/car/tests/test_car_interfaces.py 
```

Current output on my workstation
```
batman:sixpilot$ pytest --durations=10 selfdrive/car/tests/test_car_interfaces.py
=============================================================== test session starts ===============================================================
platform linux -- Python 3.11.4, pytest-8.2.0, pluggy-1.5.0
Using --randomly-seed=431984017
rootdir: /dev/shm/openpilot/sixpilot
configfile: pyproject.toml
plugins: randomly-3.15.0, subtests-0.12.1, anyio-4.3.0, cov-5.0.0, xdist-3.6.1, cpp-2.5.0, hypothesis-6.47.5, mock-3.14.0, timeout-2.3.1, flaky-3.8.1, asyncio-0.23.6
asyncio: mode=Mode.STRICT
32 workers [209 items]
........................................................................................................................................... [ 65%]
......................................................................                                                                      [100%]
============================================================== slowest 10 durations ===============================================================
2.17s call     selfdrive/car/tests/test_car_interfaces.py::TestCarInterfaces::test_car_interfaces_027_FORD_FOCUS_MK4
2.17s call     selfdrive/car/tests/test_car_interfaces.py::TestCarInterfaces::test_car_interfaces_138_MAZDA_CX9_2021
2.15s call     selfdrive/car/tests/test_car_interfaces.py::TestCarInterfaces::test_car_interfaces_037_GENESIS_GV60_EV_1ST_GEN
2.05s call     selfdrive/car/tests/test_car_interfaces.py::TestCarInterfaces::test_car_interfaces_147_SKODA_FABIA_MK4
1.97s call     selfdrive/car/tests/test_car_interfaces.py::TestCarInterfaces::test_car_interfaces_094_HYUNDAI_VELOSTER
1.97s call     selfdrive/car/tests/test_car_interfaces.py::TestCarInterfaces::test_car_interfaces_005_AUDI_Q3_MK2
1.87s call     selfdrive/car/tests/test_car_interfaces.py::TestCarInterfaces::test_car_interfaces_059_HONDA_PILOT
1.83s call     selfdrive/car/tests/test_car_interfaces.py::TestCarInterfaces::test_car_interfaces_113_KIA_OPTIMA_H_G4_FL
1.83s call     selfdrive/car/tests/test_car_interfaces.py::TestCarInterfaces::test_car_interfaces_076_HYUNDAI_IONIQ_PHEV_2019
1.78s call     selfdrive/car/tests/test_car_interfaces.py::TestCarInterfaces::test_car_interfaces_095_JEEP_GRAND_CHEROKEE
============================================================== 209 passed in 13.32s ===============================================================
```

sshane, 2024-05-28:
A good place to optimize is the hypothesis example generation. Speeds are ~2x faster than our current version in version 4.51.0 and get slower somewhere between then and 4.55.3. The latest version is even slower compared to our current version.

deanlee, 2024-06-04:
Another option is to optimize get_fuzzy_car_interface_args by caching params_strategy to avoid regenerating strategies each time. Additionally, drawing st.dictionaries is quite slow; according to my tests, it can take up to 0.3 seconds.

lsrugo, 2024-06-06:
> Another option is to optimize get_fuzzy_car_interface_args by caching params_strategy to avoid regenerating strategies each time.

I tried this and it didn't seem to work.

>  Additionally, drawing st.dictionaries is quite slow; according to my tests, it can take up to 0.3 seconds.

Right so it's likely a problem in the underlying library

geeth345, 2024-08-19:
Is this bounty still available?

Pious1020, 2024-09-14:
@adeebshihadeh, is this still open?

c-gamble, 2024-09-30:
I'm interested in working on this but it's looking like this bounty has been (partially) solved? Is the bounty still open to go from~0.7s to <= 0.2s? 

[image]

sshane, 2024-09-30:
Getting this result on a `AMD Ryzen Threadripper PRO 3955WX 16-Cores`:

```
Using --randomly-seed=4229886755
rootdir: /home/batman/openpilot
configfile: pyproject.toml
plugins: timeout-2.3.1, xdist-3.6.1, cov-5.0.0, mock-3.14.0, asyncio-0.24.0, sugar-1.0.0, randomly-3.15.0, cpp-2.6.0, flaky-3.8.1, hypothesis-6.47.5, anyio-4.4.0, subtests-0.13.1, repeat-0.9.3
asyncio: mode=Mode.STRICT, default_loop_scope=function
16 workers [211 items]    collecting ... 

 selfdrive/car/tests/test_car_interfaces.py ✓✓✓✓✓✓✓✓✓✓✓✓✓✓✓✓✓✓✓✓✓✓✓✓✓✓✓✓✓✓✓✓✓✓✓✓✓✓✓✓✓✓✓✓✓✓✓✓✓✓✓✓✓✓✓✓✓✓✓✓✓✓✓✓✓✓✓✓✓✓✓✓✓✓✓✓✓✓✓✓✓✓✓✓✓✓✓✓✓✓✓✓✓✓✓✓✓✓✓✓✓✓✓✓✓✓✓✓✓✓✓✓✓✓✓✓✓✓✓✓✓✓✓✓✓✓ 60% ██████    
                                            ✓✓✓✓✓✓✓✓✓✓✓✓✓✓✓✓✓✓✓✓✓✓✓✓✓✓✓✓✓✓✓✓✓✓✓✓✓✓✓✓✓✓✓✓✓✓✓✓✓✓✓✓✓✓✓✓✓✓✓✓✓✓✓✓✓✓✓✓✓✓✓✓✓✓✓✓✓✓✓✓✓✓✓✓✓                                         100% ██████████
================================================================================= slowest 10 durations ==================================================================================
1.27s call     selfdrive/car/tests/test_car_interfaces.py::TestCarInterfaces::test_car_interfaces_001_ACURA_RDX
1.22s call     selfdrive/car/tests/test_car_interfaces.py::TestCarInterfaces::test_car_interfaces_032_FORD_MUSTANG_MACH_E_MK1
1.20s call     selfdrive/car/tests/test_car_interfaces.py::TestCarInterfaces::test_car_interfaces_027_FORD_EXPLORER_MK6
1.17s call     selfdrive/car/tests/test_car_interfaces.py::TestCarInterfaces::test_car_interfaces_026_FORD_ESCAPE_MK4
1.16s call     selfdrive/car/tests/test_car_interfaces.py::TestCarInterfaces::test_car_interfaces_028_FORD_FOCUS_MK4
1.16s call     selfdrive/car/tests/test_car_interfaces.py::TestCarInterfaces::test_car_interfaces_042_GENESIS_GV80
1.15s call     selfdrive/car/tests/test_car_interfaces.py::TestCarInterfaces::test_car_interfaces_017_CHEVROLET_VOLT
1.14s call     selfdrive/car/tests/test_car_interfaces.py::TestCarInterfaces::test_car_interfaces_047_HONDA_CIVIC_2022
1.12s call     selfdrive/car/tests/test_car_interfaces.py::TestCarInterfaces::test_car_interfaces_025_FORD_BRONCO_SPORT_MK1
1.12s call     selfdrive/car/tests/test_car_interfaces.py::TestCarInterfaces::test_car_interfaces_202_VOLKSWAGEN_POLO_MK6

Results (14.30s):
     211 passed
batman@workstation-shane:~/openpilot$
```

So to achieve the original goal, we need to see about a ~5-6x speedup still.

c-gamble, 2024-09-30:
> Getting this result on a `AMD Ryzen Threadripper PRO 3955WX 16-Cores`:

I ran on m3 MBP—appreciate the clarification!

JIBSIL, 2025-01-08:
Hey guys,

This bounty appears open under Projects so I attempted it. I believe with a few changes I have improved it to your specifications without making the tests less effective (about a 8x improvement on my machine, i7-7700k). I'll open a PR in a bit, can you guys lock the bounty if my PR is satisfactory? First PR I've done on this repository by the way, so please let me know if there's anything I am missing.

[image]
[image]

sshane, 2025-01-08:
You can put up a PR and we will review it.

blackoutnet, 2025-05-14:
Hey guys, please have a look: https://github.com/commaai/openpilot/pull/35183

iam-saju, 2025-06-18:
is it still available

hanzhi227, 2025-07-09:
There's 227 tests, so if each test average is <0.2s then the total test time <45.4s which it is already at ~22s for me. Simplest way to reduce the test time so that the highest test times are around 0.2s is to just reduce the MAX_EXAMPLES from 60 -> 15, but even 1t 60, the test times is averaging <0.1s per test.

`hanzhi$ MAX_EXAMPLES=15 pytest --durations=10 --cache-clear selfdrive/car/tests/test_car_interfaces.py 
================================================================================================================ test session starts ================================================================================================================
platform darwin -- Python 3.11.4, pytest-8.4.1, pluggy-1.6.0
Using --randomly-seed=1967957443
rootdir: /Users/hanzh/Dev/Projects/openpilot
configfile: pyproject.toml
plugins: randomly-3.16.0, cpp-2.6.0, repeat-0.9.4, timeout-2.4.0, mock-3.14.1, hypothesis-6.47.5, xdist-3.7.1.dev3+g909e97b, subtests-0.14.2, asyncio-1.0.0
asyncio: mode=Mode.STRICT, asyncio_default_fixture_loop_scope=function, asyncio_default_test_loop_scope=function
8 workers [227 items]   
...................................................................................................................................................................................................................................           [100%]
=============================================================================================================== slowest 10 durations ================================================================================================================
0.28s call     selfdrive/car/tests/test_car_interfaces.py::TestCarInterfaces::test_car_interfaces_028_FORD_ESCAPE_MK4
0.28s call     selfdrive/car/tests/test_car_interfaces.py::TestCarInterfaces::test_car_interfaces_029_FORD_ESCAPE_MK4_5
0.28s call     selfdrive/car/tests/test_car_interfaces.py::TestCarInterfaces::test_car_interfaces_035_FORD_MAVERICK_MK1
0.26s call     selfdrive/car/tests/test_car_interfaces.py::TestCarInterfaces::test_car_interfaces_031_FORD_EXPLORER_MK6
0.26s call     selfdrive/car/tests/test_car_interfaces.py::TestCarInterfaces::test_car_interfaces_032_FORD_FOCUS_MK4
0.25s call     selfdrive/car/tests/test_car_interfaces.py::TestCarInterfaces::test_car_interfaces_033_FORD_F_150_LIGHTNING_MK1
0.24s call     selfdrive/car/tests/test_car_interfaces.py::TestCarInterfaces::test_car_interfaces_030_FORD_EXPEDITION_MK4
0.23s call     selfdrive/car/tests/test_car_interfaces.py::TestCarInterfaces::test_car_interfaces_027_FORD_BRONCO_SPORT_MK1
0.18s call     selfdrive/car/tests/test_car_interfaces.py::TestCarInterfaces::test_car_interfaces_000_ACURA_ILX
0.18s call     selfdrive/car/tests/test_car_interfaces.py::TestCarInterfaces::test_car_interfaces_037_FORD_RANGER_MK2
================================================================================================================ 227 passed in 6.42s ================================================================================================================`

TheSecurityDev, 2025-11-09:
> The above pr needs to be merged in opendbc repo to see performance improvements in the main openpilot repo. Some eyes here when you get a chance please.

I believe you can test this by forking both openpilot and opendbc, and configuring your openpilot fork to point to your opendbc fork.

TheSecurityDev, 2025-11-09:
> will those tests still run? I have no changes for openpilot repo.

I think it will

logesh45, 2025-11-20:
https://github.com/commaai/opendbc/pull/2888#pullrequestreview-3490056013

@adeebshihadeh @sshane @jyoung8607 can this be marked as solved now? I'm looking for an interview slot.

FELMONON, 2026-01-22:
Created PR for this: https://github.com/commaai/opendbc/pull/3052

The fix caches hypothesis strategies at module level instead of recreating them on every test call. Strategy creation is expensive, so this should significantly speed up test execution.

rob0tics, 2026-02-03:
I've submitted a PR to opendbc that fixes the collection speed issue. It reduces collection time from ~20s to 0.18s.

PR is here: [[Link to your opendbc PR]](https://github.com/commaai/opendbc/pull/3107)

javery556, 2026-03-15:
I opened the related opendbc PR here:
https://github.com/commaai/opendbc/pull/3212

This keeps the test coverage intact and targets the initialization-time hotspots I found on current `master`:
- cache `CANDefine` values per DBC
- precompute the constant speed-KF gain used by `CarStateBase`

Local benchmarks on the same machine:
- full xdist run: `241 passed in 7.77s` -> `241 passed in 6.42s`
- full single-process run after the patch: `241 passed in 46.94s`, slowest test `0.94s` and average about `0.195s/test`

If there is still payable scope on this bounty after the earlier January opendbc merge, this should be ready for review.