commaai/openpilot #32543: GM Cam ACC : Distance button not properly handling dash menu

niknak6, 2024-05-26:
### Describe the bug

Description: When the distance button on the steering wheel is pressed, it displays a menu on the car's dash showing the current set distance. Subsequent presses should cycle through the distance options (Near, Medium, Far). However, OpenPilot is incorrectly interpreting the initial button press, which only opens the menu on the dash, as a command to cycle the personality setting. This results in an unintended change in the personality within OpenPilot upon the first press, rather than simply opening the menu and registering subsequent presses, keeping the dash and OpenPilot in sync with one another.

Near-> Aggressive
Medium  -> Standard
Far -> Relaxed

Steps to Reproduce:

- Press the distance button on the steering wheel to open the distance menu on the dash.
- Observe the displayed distance setting.
- Despite the distance on the dash not changing and only opening the menu, the personality within OpenPilot will change. If the dash was set to medium and OpenPilot's personality was set to standard, OpenPilot would now be at Relaxed while the GM dash would still be on medium, now being out of sync.

Expected Behavior: The first press of the distance button should only open the menu displaying the current distance setting. Subsequent presses should cycle through the distance options (Near, Medium, Far) should the user press the distance button again while the menu is open.

Proposed Solution: For GM cam cars, the initial press of the distance button should not trigger an OpenPilot personality change. Only subsequent presses within x time frame (x being however long the menu takes to timeout on dash) of the first press should change OpenPilot personalities.

I can confirm this issue exists on upstream supported Bolts and 1500 Silverados at this time. 

Attaching video link to show OpenPilot personality changing when menu is opened but dash distance isn’t changed. 

[](https://imgur.com/d7BOxAy)

### Provide a route where the issue occurs

b3ece9b5a1d3488c/00000001--fe9506f08d/0

### openpilot version

0.9.7

niknak6, 2024-05-26:
[link to video](https://imgur.com/a/QCr8eQe)

morrislee, 2024-05-27:
I believe you are describing known issue here #31911

niknak6, 2024-05-27:
> I believe you are describing known issue here #31911 

Hi there!

The issue referenced would be the reason it's not syncing the personality back to the dash. As a result of this, we'd need special handling for the dash menu. 

Example:

On first button press, don't change personality within OpenPilot. Only change personality if there are subsequent button presses within x seconds (while the menu is open), keeping the dash and OpenPilot personality in sync and properly handling the menu.

morrislee, 2024-05-27:
> Hi there!
>
> The issue referenced would be the reason it's not syncing the personality back to the dash. As a result of this, we'd need special handling for the dash menu.
>
> Example:
>
> On first button [...]

I dont believe a hard coded delay will be an acceptable fix, I think we are unable to read the current set following distance. If we can check the current set distance is a better solution.

niknak6, 2024-05-27:
> I dont believe a hard coded delay will be an acceptable fix, I think we are unable to read the current set following distance. If we can check the current set distance is a better solution.

While I typically agree that finding a solution to the issue you referenced would be more acceptable and certainly more clean, if the car will not allow for it in most scenarios, the alternative option is what we currently have... out of dash/OpenPilot personality. We had this working in forkland well before upstream gm distance button support, but it could only stand to benefit everyone.

niknak6, 2024-05-27:
Here is how we did exactly this prior to upstream comma distance button support. 

https://github.com/FrogAi/FrogPilot/blob/FrogPilot-Old/selfdrive/car/gm/carstate.py#L164

morrislee, 2024-05-27:
> > > > I believe you are describing known issue here #31911
> > 
> > 
> > > 
> > 
> > 
> > > Hi there!
> > 
> > 
> > > 
> > 
> > 
> > > The issue referenced would be the reason it's not syncing the personality back to the dash. As a result of this, we'd need special handling for the dash menu.
> > 
> > 
> > > 
> > 
> > 
> > > Example:
> > 
> > 
> > > 
> > 
> > 
> > > On first button press, don't change personality within OpenPilot. Only change personality if there are subsequent button presses within x seconds (while the menu is open), keeping the dash and OpenPilot personality in sync and properly handling the menu.
> > 
> > 
> > I dont believe a hard coded delay will be an acceptable fix, I think we are unable to read the current set following distance. If we can check the current set distance is a better solution.
> 
> While I typically agree that finding a solution to the issue you referenced would be more acceptable and certainly more clean, if the car will not allow for it in most scenarios, the alternative option is what we currently have... out of sync dash/OpenPilot personality. We had this working in forkland well before upstream gm distance button support, but it could only stand to benefit everyone.

I absolutely agree, but I am only interpreting what comma may say. If the solution is solid, perhaps make a PR?

morrislee, 2024-05-27:
@sshane, would you kindly take a look at this if this will be accepted fix?

FrogAi, 2024-05-27:
The tl;dr of the "fix" was this:

1. User clicks the distance button
2. If they click it again within 3.5 seconds, it changes the personality profile and it resets the 3.5-second timer
3. If the 3.5-second timer expires, it doesn't change anything

sshane, 2024-05-27:
Not sure if that works because the two could still be out of sync when the car starts up.

fredyshox, 2024-05-27:
Duplicate of https://github.com/commaai/openpilot/issues/31911

closed 2024-05-27 by fredyshox

niknak6, 2024-05-27:
> Not sure if that works because the two could still be out of sync when the car starts up. 

You're 100 percent correct. The user would have to manually put them in sync the first time. But from that point, the car and OpenPilot remember the personality last used. But that's a one time set and forget.

morrislee, 2024-05-27:
if what @niknak6 say its true, it is far better than the guessing game we have now. Perhaps we will never have a way to read the state

sshane, 2024-05-27:
That issue should have a way to handle this, someone just needs to write it and try it.