commaai/openpilot #32553: New Ubuntu setup flow

adeebshihadeh, 2024-05-27:
We'll offer two paths:
1. manually install/setup everything, for those who like to manage their own env and tooling
2. an interactive script that'll get you setup from scratch

The interactive script will go something like this:
* do you want to install these runtime apt dependencies?
* do you want to install apt development dependencies?
* do you want to install apt nice-to-haves (e.g. valgrind)?
* detect minimum python version
* start a venv
* install python dependencies inside venv
* show how to active venv

am2024ma, 2024-05-27:
Is it OK to test in virtual machine Ubuntu latest LTS.

navistonks, 2024-05-30:
What should be the optional nice-to-have packages?
And isn't the current script already using venvs, since poetry uses them?

adeebshihadeh, 2024-05-30:
> What should be the optional nice-to-have packages?

The important thing is to get the groups setup, then we can sort everything out later. Just find at least one to move to nice to have. 

> And isn't the current script already using venvs, since poetry uses them?

Yes, but the script should explain what it's going to do and ask if you want to do it. In this case, some users may not want to install poetry and just want to `pip install .`.

closed 2024-06-28 by adeebshihadeh