commaai/openpilot #32660: [$100 bounty] Remove `setproctitle` dependency

adeebshihadeh, 2024-06-08:
Replace `setproctitle` with our own small helper in `common/`. Just needs to work on Linux.

markypizz, 2024-06-08:
Curious if I'm trending in the right direction. Thanks!
https://github.com/commaai/openpilot/pull/32666

jyoung8607, 2024-06-10:
Truly replacing Linux setproctitle() requires clobbering the original argv(), which is some real sausage-making. However, @cl0cks4fe has taken a shot at #32691.

The other approaches are setting a thread title. `htop` doesn't show them because [it's configured not to](https://github.com/commaai/openpilot/pull/32667#issuecomment-2159110183). IMO we should fix the htop config, because then we'll actually start seeing see the thread names we're already setting in camerad, loggerd, and boardd. Then we can migrate setproctitle() users to setting thread titles, and dispose of the dependency that way.

markypizz, 2024-06-11:
Agreed with the above. htop is looking at /proc/{pid}/cmdline, which can't be changed other than clobbering argv and the mess associated with it. Is the setting of the title purely for cosmetics or for some other reason?

PR #32667 (2024-06-11): Custom setproctitle
Resolves: #32660  

It's a simple refactor with a test to confirm correctness.
closed 2024-06-11 by adeebshihadeh