commaai/openpilot #32669: Cruise Fault when OP Long is enabled

bobbygg, 2024-06-09:
### Describe the bug

When I enable OP Long before starting the car, I always get the cruise fault error. I've restarted the car many times as instructed and still get the same error. I've also tried reinstalling the release version, master-ci, SP, and FP. I get the same error each and every time. However, enabling OP Long after starting the vehicle has no error but it also doesn't do anything either. I just upgraded to the C3X from a C2.

### Which car does this affect?

Honda CRV Hybrid 2021

### Provide a route where the issue occurs

4db6f59e9b7a842f/00000001--dfb48bf9f0

### openpilot version

0.9.7

### Additional info

I noticed my car was hugging the right side of the lane during the following test drive too:
4db6f59e9b7a842f/00000004--e296d70b02

sshane, 2024-06-10:
Did alpha longitudinal used to work on your previous device?

bobbygg, 2024-06-10:
> Did alpha longitudinal used to work on your previous device or is this your first time trying it?

No. This is the first time I'm using it. I just upgraded to the C3X from C2 a couple of weeks ago.