commaai/openpilot #32717: [$500 bounty] eSIM provisioning on comma 3X

adeebshihadeh, 2024-06-12:
comma 3X ships with an eSIM that comes provisioned from the factory for [comma prime](https://comma.ai/connect). The goal of this bounty is to come up with a solution for users to provision them with their own service, which we can eventually integrate into the device settings or [comma connect](https://connect.comma.ai).

The deliverable is a Python script that sets up an arbitrary eSIM profile on the comma 3X. The input is the eSIM activation code (e.g. `LPA:1$smdp.io$K2-1V5B9M-DD9VC9`).

Notes
* comma 3X uses the [Quectel EG25](https://www.quectel.com/product/lte-eg25-g/)
* comma 3X comes with an [SGP 32](https://www.gsma.com/solutions-and-impact/technologies/esim/gsma_resources/sgp-32-v1-0/) compliant eSIM
* the SIM can also be ordered separately from [here](https://comma.ai/shop/comma-prime-sim) for $20
* may be possible to do this whole bounty on something other than a comma 3X; we just care if it works on a 3X

The EG25 has special firmware with this capability (see [esim.py](https://github.com/commaai/openpilot/blob/master/system/hardware/tici/esim.py)), however the goal of this bounty is to do it without that firmware.

closed 2024-08-15 by adeebshihadeh

andiradulescu, 2024-08-15:
What was the conclusion?

BBBmau, 2024-08-15:
interested in hearing what led to completion also.

adeebshihadeh, 2024-08-15:
Got a profile loaded with https://github.com/estkme-group/lpac