commaai/openpilot #32737: HKG: stock ACC radar fault causing unexpected disengage with no warning from openpilot

BBBmau, 2024-06-13:
### Describe the bug

after some investigation from @jyoung8607 and @sunnyhaibin, thanks by the way!

From discord:
`
@jyoung8607: Looks like a brief fault from your stock ACC radar, one that openpilot doesn't currently handle very well. Stock ACC briefly goes to a possibly-faulted state that we don't directly handle. The drivetrain side doesn't fault, which sucks because that's the side openpilot does handle. Stock ACC subsequently disengages in what the code thinks is a normal way, but there's no external triggers like brake or button presses that I'm seeing. 
`

`
@sunnyhaibinSkimmed: through some signals quickly (have to head out shortly), TCS13|ACCEnable stayed healthy, so the ESC wasn't throwing a DTC here. Personally I have never seen this case before. 
`

[image]
[image]

### Which car does this affect?

Hyundai Sontata 2023

### Provide a route where the issue occurs

9a31d18cd1a79c69/00000004--ceb1d98f38/61

### openpilot version

0.9.7-staging

### Additional info

link to the video segment: https://connect.comma.ai/9a31d18cd1a79c69/1718290279773/1718290350415

sshane, 2024-06-13:
Duplicate of https://github.com/commaai/openpilot/issues/23601

[image]

sshane, 2024-06-13:
This happened later in the segment, both rising while engaging, putting the car in some sort of soft disable state.

For now, I think we can safety mark TCS events as accFaults to properly alert here, since this is fairly common.

@jyoung8607 @sunnyhaibin any of you want to make a PR for this?

BBBmau, 2024-06-13:
I'd like to pick this up, been really wanting to find an issue to be able to work with cabana and understand the car states more.

jyoung8607, 2024-07-01:
This specific issue was resolved in #32743, but #23601 should remain open to detect this more generally.

closed 2024-07-01 by jyoung8607